capacitor-community/bluetooth-le #4: implement getDevices() for android and ios

jrobeson, 2020-12-18:
**Is your feature request related to a problem? Please describe.**
It doesn't seem possible to get multiple devices to interact with them. 
`requestDevice()` will never return a single result for my usecases.

**Describe the solution you'd like**
support `getDevices()` from the web bluetooth spec

pwespi, 2020-12-18:
Thank you for using this plugin and for submitting this feature request.

It is already possible to call `requestDevice()` more than once and store the results, then you can interact with several devices at the same time.

But I assume you would prefer a single `scan` function (or similar) and get all discovered devices in a callback? Would you mind describing your use case in more detail? This would help me design this feature. Thanks.

jrobeson, 2020-12-18:
You can? The Examples weren't clear about that, but how does it work if you have multiple devices with the same name offering the same services? The spec didn't seem to clearly say how you'd disambiguate in that situation.  There'd no such ambiguity by implementing `getDevices()`

pwespi, 2020-12-28:
I released a new version of the plugin which has [requestLEScan](https://github.com/capacitor-community/bluetooth-le#requestlescan) and [stopLEScan](https://github.com/capacitor-community/bluetooth-le#stoplescan).

closed 2020-12-28 by pwespi