capacitor-community/bluetooth-le #5: Some useful features not found in the web bluetooth spec

jrobeson, 2020-12-18:
**Is your feature request related to a problem? Please describe.**
Some useful features found in other common bluetooth plugins/modules.

This plugin seems most promising for the future as it uses modern android (at least, can't speak for ios) features and seems relatively well designed. I might be wanting too much from it though.

 * reporting duplicate advertisements - boolean in scan filters to allow reporting the same device as many times as it advertises
 * scan timeout - ability to restrict the time spent scanning
 * bonding - but maybe i missed that when reading the spec

Anything else that might seem useful from: https://github.com/randdusing/cordova-plugin-bluetoothle

jrobeson, 2020-12-18:
Sometimes I have 20+ BLE devices that I need data from (and quickly). and due to device connection limits I can't keep a constant connection to all of them. I instead send the the data in the advertisement packets.

pwespi, 2020-12-28:
* [x] requestLEScan
* [x] stopLEScan
* [x] allowDuplicates
* [ ] bonding

tellybrown, 2021-01-21:
some sort of listener promise to be notified when bluetooth is turned on/off to facilitate an event driven connect/disconnect of devices  TIA

BillBu, 2021-02-12:
Thanks for creating this plugin - very clean to use. 
Registering for disconnect events as per @tellybrown's comment above would be a most welcome addition. Use case for me is when devices go out of range or are turned off.

pwespi, 2021-02-13:
@tellybrown @BillBu Thanks for your feedback. I just released v0.2.0 with the following features:
* `getEnabled`
* `startEnabledNotifications`
* `stopEnabledNotifications`
* optional `onDisconnect` callback for `connect` 

I hope this will be useful.

BillBu, 2021-02-13:
That's fantastic @pwespi thank you so much. I'll check it out this week.

kegtron, 2021-02-17:
> @tellybrown @BillBu Thanks for your feedback. I just released v0.2.0 with the following features:

Sounds like a great addition. Can you create a v0.2.0 release tag?

Thanks!

pwespi, 2021-02-17:
> 
> Sounds like a great addition. Can you create a v0.2.0 release tag?
> 
> Thanks!

Done. It was already published on npm, but I forgot to push the git tag. Thanks for letting me know.

pwespi, 2021-02-26:
Thank you for all your feedback.

I've created a new issue for bonding #50.

I think all the other points mentioned here have been addressed, so I'm closing this issue. If you have any other bugs or feature requests, feel free to open new issues.

closed 2021-02-26 by pwespi