capacitor-community/bluetooth-le #12: 'capacitorExports is not defined' error during bundling

spaghettiguru, 2021-01-07:
First of all, thank you for your work on the plugin @pwespi, I believe it will help us greatly when we make it work :)

Here is the issue:

**Describe the bug**
The web assets build is failing if the following `import { BleClient } from '@capacitor-community/bluetooth-le'; `defined somewhere in the code.

The error is the following:
> ReferenceError: capacitorExports is not defined
>     at Object.<anonymous> (/projectDir/node_modules/@capacitor-community/bluetooth-le/dist/plugin.js:403:7)

The contents of the plugin.js file at line 403 (with some additional lines above, for context):

```
   ...

   exports.BleClient = BleClient;
    exports.BluetoothLe = BluetoothLe;
    exports.BluetoothLeWeb = BluetoothLeWeb;
    exports.dataViewToHexString = dataViewToHexString;
    exports.dataViewToNumbers = dataViewToNumbers;
    exports.dataViewToText = dataViewToText;
    exports.hexStringToDataView = hexStringToDataView;
    exports.mapToObject = mapToObject;
    exports.numberToUUID = numberToUUID;
    exports.numbersToDataView = numbersToDataView;
    exports.textToDataView = textToDataView;
    exports.webUUIDToString = webUUIDToString;

 Object.defineProperty(exports, '__esModule', { value: true });

 return exports;

}({}, capacitorExports)); // <-- line 403 from the error
```

We are using the NextJS's `next build` command to run build, but I guess it doesn't matter which bundler is used.

**To Reproduce**
Steps to reproduce the behavior:
1. install bluetooth-le capacitor plugin and import it somewhere.
2. Run the build to bundle your app.

**Expected behavior**
The build is passing without errors from the plugin.

codingtub, 2021-01-07:
I have exactly the same problem, I'm on next v10.0.4.
I have to say for me its the first time trying this plugin on Next, so I unfortunately can't tell if it was working for any prior versions.

pwespi, 2021-01-07:
Thank you for trying out this plugin and for opening this issue. 
I can reproduce it locally with this repo: https://github.com/pwespi/nextjs-test and will look into it.

pwespi, 2021-01-07:
Probably related to https://github.com/ionic-team/capacitor/issues/3972

pwespi, 2021-01-08:
I have found a possible solution. See #14.

You can test it with `npm i @capacitor-community/bluetooth-le@0.1.0-15-ge63b6a9` and change the import to `import { BleClient } from '@capacitor-community/bluetooth-le/dist';`.

(I'm not yet sure if it's the best solution, so this might change again.)

pwespi, 2021-01-09:
Closed in #14 and released as 0.1.1.

(no need to change the import)

closed 2021-01-09 by pwespi

spaghettiguru, 2021-01-10:
Thank you very much for help @pwespi, you saved me from messing about with webpack configuration :)

codingtub, 2021-01-11:
Thanks a lot, everything's working smoothly now! 🎉