capacitor-community/bluetooth-le #32: How to test in the browser using ionic serve

baxeico, 2021-01-28:
Sorry if I open an issue here. That is a question:

I'm experimenting with your plugin in an Ionic project. It could be very useful to test Bluetooth functionalities directly from my development laptop (which of course has bluetooth), from the browser. I mean starting `ionic serve` and going to `http://localhost:8100/`. I'm asking because I read that the plugin support Web Bluetooth and I'm on Chrome that should support this API.

Do you think it's possible or I have to test on a real Android device?

Thank you very much and keep up the good work!

pwespi, 2021-01-28:
Thanks for using this plugin.

Yes, testing your app with this plugin in Chrome is possible and should work just out of the box when running `ionic serve`.

There are a few things to note about Web Bluetooth (in general, not just in this plugin):
* `requestDevice` "must be handling a user gesture to show a permission request". This means you have to use it e.g. in a click handler of a button, and probably cannot use it in something like `ngOnInit`.
* `requestLEScan` is still experimental and behind a flag in Chrome, see [here](https://github.com/WebBluetoothCG/web-bluetooth/blob/gh-pages/implementation-status.md#scanning-api). Also it has a [bug](https://bugs.chromium.org/p/chromium/issues/detail?id=1137504) that makes it hard to use even if you enable the flag.

If you encounter any errors or problems, let me know.

baxeico, 2021-01-28:
Thank you very much! I needed to enable the "Experimental Web Platform features" in chrome:flags.
Now it's working!

closed 2021-01-28 by baxeico