capacitor-community/bluetooth-le #49: Write without response

maxarndt, 2021-02-25:
Hey @pwespi!

Great work with this plugin, thank you very much!
I really like the clean API of this plugin. Connecting to a device and reading characteristics works perfectly on my iOS devices as well as with Google Chrome.

Unfortunately I failed to write to a characteristic with an iOS device. It works well with Google Chrome but on iOS i got the following error:
```
Reject write|SOME_SERVICE_UUID|SOME_CHARACTERISTIC_UUID Writing is not permitted.
```

After a while of debugging I found the cause for my issue. The `.write()` function exposed by this plugin always uses `.withResponse` as type in the call below:

https://github.com/capacitor-community/bluetooth-le/blob/024aca0bd289c4c96d7d657f66db9d911d30ef0f/ios/Plugin/Device.swift#L118

Unfortunately my characteristic only allows write without response. 

Could we get such write operations supported on iOS as well? That would be awesome!

pwespi, 2021-02-26:
Thanks for using this plugin and for the detailed issue.

I will consider this for the next version.

closed 2021-02-27 by pwespi

pwespi, 2021-02-27:
`writeWithoutResponse` has been added in version v0.3.0.

PR #53 (2021-02-27): feat(): add writeWithoutResponse
closes #49