capacitor-community/bluetooth-le #60: Expose localName of advertising packets in ScanResult object

maxarndt, 2021-03-04:
**Is your feature request related to a problem? Please describe.**
When a BLE peripheral is discovered the first time by an iOS device the `localName` from the advertising packet is shown to the user as device name.
After connecting to this device the GAP name is discovered and cached. 
For subsequent connection attempts the cached GAP name is shown to the user instead of the `localName` in the advertising packet.

If you are using the `requestDevice(...)` function of this plugin a dialog is opened automatically which shows eventually cached names. 

If you have a BLE device where the GAP name differs from the `localName` provided in the advertising packets you can not guarantee a proper user experience using the `requestDevice(...)` function.

By using the `requestLEScan(...)` function you could provide a proper user experience by fetching the `localName` from the advertising packet of each found device and render your own connection dialog.
Unfortunately the `localName` attribute is currently not accessible in the [ScanResult](https://github.com/capacitor-community/bluetooth-le#scanresult) object.

**Describe the solution you'd like**
Add the `localName` property from the advertising packet to the [ScanResult](https://github.com/capacitor-community/bluetooth-le#scanresult) object.

**Additional context**
I tested it out by adding the following snippet to the `getScanResult()` function of the `Plugin.swift` file right before the `return` Statement:

```swift
let localName = advertisementData[CBAdvertisementDataLocalNameKey] as? String
if localName != nil {
    data["localName"] = localName
}
```

But I am not sure if `localName` is a good attribute name next to the `device.name` which is already present in the `ScanResult`. Personally, I would prefer a name that makes it clear that it is the name from the advertisement packet.

Do you think this is something you could add to this plugin? 
Thank you in advance for your time and efforts!

pwespi, 2021-03-07:
Thanks for this issue.

I can definitely add this. However, I'm not sure what to do on Android and web. Actually, the caching behavior you're describing only exist on iOS, as far as I can tell. So there might be an inconsistent behavior of `device.name` across platforms. Are you using the plugin only on iOS or also on the other platforms? 

Possible options:
* only add it to iOS
* add it to all platforms, but on Android and web it will always be equal to `device.name`

Any thoughts on this?

What do you think about calling it `advertisementName`?

Out of interest, what kind of devices use a different name in the advertisement than the GAP name?

maxarndt, 2021-03-08:
Hey @pwespi! Thank you for your quick reply!

Currently I'm using your plugin only on iOS but later this day I should receive an Android phone to further evaluate our prototyping. 

Are there any problems by adding the `localName` of the advertisement packet for Android and the web platform? 
If it is possible to access this value on these platforms I would add it as well.
I currently don't see a reason why it should not be added to the Android and web platform or even be set to the value of `device.name`.
A BLE device can have both, the GAP name and the advertisement localName so I would provide both and each developer using this plugin can choose which one to use.

I've thought about the name again and would currently prefer `localName`. 
That's the name which is used in the spec and the name is commonly used across articles, Q&A sites and blogs.
So I believe that is the name what people would expect here.

We have some commercial IoT devices here where I stumbled across this issue. 
But those are not consumer products which can currently be found somewhere on the web.

The Bluetooth [Core Specification Supplement 9](https://www.bluetooth.com/specifications/specs/) spec says in section 1.12 the following:
> The Local Name data type shall be the same as, or a shortened version of, the local name assigned to the device. The Local Name data type value indicates if the name is complete or shortened. If the name is shortened, the complete name can be read using the remote name request procedure over BR/EDR or by reading the device name characteristic after the connection has been established using GATT.  
   
So what we have encountered on our devices here does not meet the spec.
But because of no one can guarantee that the names are always identically I think having both properties available in the `ScanResult` object would be ideal.

pwespi, 2021-03-08:
Thanks for all the information. 

There is no problem to add a `localName` property in Android and web as well (the point is, I think on those platforms `device.name` is already the localName). So I will probably add `localName` to all platforms and document that on iOS `device.name` is the cached GAP name after connecting.

Since the advertisement data only contains the localName, there is no way to know the GAP name while scanning without caching it from previous connections (correct me if I'm wrong). Apparently iOS does this under the hood. I could try to add GAP name caching to Android and web, but not sure if it's a good idea. I'd rather leave that to the app if it is really required.

What do you think?

maxarndt, 2021-03-09:
Ah yes, of course!

Yesterday I lost sight of the fact that the `device.name` attribute on Android and the web platform already contains the `localName` from the advertisement packets. 

Probably I have been confused because of the current naming of `device.name`. I assumed this field shows the content of the GAP Device Name. But without caching there is no way to include the GAP Device Name in the `ScanResult` because the GAP Device Name characteristic can of course only be read after a connection has been established.

I fully agree with you, I would not implement caching in this plugin. That would bring a high level of complexity and behavior that users of this plugin would probably not expect.

---

> So I will probably add `localName` to all platforms and document that on iOS `device.name` is the cached GAP name after connecting.

I think this is currently the best solution. For Android and Web this means that `localName` and `device.name` should always be identical, but the resulting disadvantage of a somewhat larger response should have virtually no effect in this environment. From my point of view, the predominant advantage is that both the `localName` and `device.name` fields are available on all three platforms. As a result, when developing a multi-platform app, it does not have to be taken into account that, depending on the platform, an attribute is available or not.

Do you see any other drawbacks that would come with this solution?

pwespi, 2021-03-09:
I agree that the larger response doesn't really matter and that it's best if both `localName` and `device.name` are there on all platforms. I don't see any other drawbacks and will go with that solution.

Thanks for the fruitful discussion.

pwespi, 2021-03-14:
This has been added to v0.4.0 (Capacitor 2) and v1.0.0-1 (Capacitor 3).

closed 2021-03-14 by pwespi