capacitor-community/bluetooth-le #63: (Capacitor 3)

dennisameling, 2021-03-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Capacitor 3 app with the `@capacitor-community/bluetooth-le` plugin installed
2. Run the `BleClient.initialize()` method on iOS 13+
3. See the request for permission. Make sure to press "Decline":

[image]

4. Notice the `BleClient.initialize()` method never finishes (the promise never resolves):

```
⚡️  To Native ->  BluetoothLe initialize 102368098
```

NOTE: when you reject once, the app will never re-request permissions. You'll have to manually enable the permission in iOS settings or reinstall the app.

**Expected behavior**

The promise should trigger a rejection, like it does on Android

**Screenshots**
See above

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser:  Chrome
 - Version Big Sur 11.2.3

**Smartphone (please complete the following information):**
 - Device: iPhone7
 - OS: iOS 14.4 (but applicable to iOS 13+)
 - Browser built-in WebView
 - Version ?

**Additional context**
Apparently in Capacitor 3, the `checkPermissions()` and `requestPermissions()` functions have to be implemented: https://capacitorjs.com/docs/v3/plugins/ios#permissions

in iOS 13+, the user has to be asked for Bluetooth permission explicitly: https://medium.com/flawless-app-stories/handling-ios-13-bluetooth-permissions-26c6a8cbb816

Let me know if I can help with the implementation of those methods!

pwespi, 2021-03-14:
Thanks for this issue. The hanging initialize() call has been fixed in v0.4.0 (Capacitor 2) and v1.0.0-1 (Capacitor 3).

Regarding `requestPermissions`, I don't think it is possible to request the Bluetooth permission from within the app on iOS once it as been denied. According to [here](https://developer.apple.com/documentation/uikit/protecting_the_user_s_privacy/requesting_access_to_protected_resources):

> When your app attempts to access a protected resource after its first attempt, the system remembers the user’s permission choice and doesn’t prompt again. To prompt the user again, you must reset access to these resources on your device or system.

So I think the best you can do is ask the user to go to the settings and enable Bluetooth there. 

Any other idea?

pwespi, 2021-03-16:
Closing this as the main issue has been solved. If there is anything you think the plugin can do to help with denied permissions, feel free to open a new one.

closed 2021-03-16 by pwespi