capacitor-community/bluetooth-le #64: Queueing

maxarndt, 2021-03-10:
**Is your feature request related to a problem? Please describe.**
By using this plugin my goal is to write my BLE business logic once and can run it on Android, iOS and maybe someday on the web. 
As I mentioned few days ago I started developing against iOS devices what worked out really well with this awesome plugin!
Two days ago I received an Android phone to evaluate this platform as well.

Unfortunately there seems to be a quite notable difference running the same codebase which uses this plugin on iOS and Android.

While reading two characteristics in parallel works without any trouble on iOS it fails on Android. 
The same happens for starting or stopping two notifications in parallel.
The underlying function rejects with `Reading characteristic failed.` and `Setting notification failed.` respectively. 

After some research I found some articles (e.g. [this one](https://punchthrough.com/android-ble-development-tips/)) complaining about BLE usage on Android compared to iOS among others because of missing queueing in the Android BLE stack.
A quick look at the Cordova BLE plugins revealed that those seem to have queueing implemented for the Android side of the plugin (see https://github.com/randdusing/cordova-plugin-bluetoothle/pull/261 and https://github.com/don/cordova-plugin-ble-central/issues/2).

**Describe the solution you'd like**
As a user of this plugin I expect that my app works on all supported platforms similarly.

I would not want to implement some kind of queuing in my capacitor app which only gets executed if running on Android.
Neither I want to implement queueing in general in my app.
I think this should not be handled in the code of the user utilizing this plugin.

Do you think this should be handled by this plugin?

**Describe alternatives you've considered**
None so far.

**Additional context**
Next to the Android platform the web platform needs such queueing as well.
We did some development with BLE enabled PWAs in the past which run on Google Chrome on Android.
We utilized the Web Bluetooth API directly and ended up implementing a queue for most of the BLE operations. 
As a side note, from my experience running such BLE enabled web apps with Chrome on macOS don't need to have queueing, it seems macOS has some kind of queueing in place as iOS does.
But of course web development should in general be handled independent of the clients operating system which means that such a queue should also be implemented there.

Out of curiosity, what are your BLE use cases and which platforms do you utilize?

pwespi, 2021-03-14:
Thank you for this issue and all the details.

I'm aware that some other plugins use queuing on Android. So far, I've been able to use the same app code on all three platforms with this plugin. My typical use case is something like:
* connect to a device
* subscribe to one or two notifications (using `await`, so one after the other)
* then use `write` to send commands to a characteristic and receive command results and data over the notifications

Since I'm awaiting all calls to the plugin and I don't allow writing a command before the result from the previous has arrived, I have an implicit queue in my app code.

However I agree with you that the plugin should behave the same on all platforms, so it would be nice if the plugin can handle the queue where needed. I'm thinking about adding a queue in the JavaScript layer of the plugin, which would be enabled by default on Android and web and disabled on iOS. I'm not sure if there would be a noticeable performance difference between having the queue in Android code and having it in JavaScript code. Any thoughts?

maxarndt, 2021-03-15:
Thank you for sharing your typical use case.

I like the idea of adding a queue in the JavaScript layer of the plugin for Android and the web platform. 
Since I assume for most use cases that the queue is likely to contain very little or no items most of the time, I would not assume any noticeable performance differences when implemented in Android code or at the JavaScript layer.

If someone reports a performance issue on Android one day the implementation could be questioned again. 
The good thing is that a JavaScript implementation is needed anyway, at least for the web platform.

pwespi, 2021-03-20:
A queue has been added to `BleClient` in v0.5.0 (for Capacitor v2) and v1.0.0-2 (for Capacitor v3). It's enabled by default on all platforms, because 1) even on iOS if you try to use a single characteristic in parallel, the plugin fails (a plugin issue, not a iOS issue) and 2) the performance difference is really small.

If you want to disable the queue, you can use `BleClient.disableQueue()`.

closed 2021-03-20 by pwespi