capacitor-community/bluetooth-le #65: startNotifications doesn't clean up after livereload

pentabarf, 2021-03-10:
**Describe the bug**
Old notifications don't stop after Ionic Capacitor livereload
Callback gets called multiple times per notification in the reloaded app.

**To Reproduce**
Steps to reproduce the behavior:
1. create ionic app with capacitor support and bluetooth-le plugin
2. create service which:
3. connect to ble device
4. start notification and logs data to console in callback
5. start app with ionic capacitor run --livereload --external
6. save project file to trigger livereload
7. app again connects to ble device and starts notification
8. all notification data received twice

**Expected behavior**
native notification listeners needs to get stopped when app is reloaded.
notification data has to receive one per time after livereload.

**Screenshots**
[image]
console gets flooded with these messages after multiple live reloads

**Desktop:**
 - OS: MacOS
 - Browser Chrome
 - Version 88.0.4324.192

**Smartphone:**
 - Device: Google Pixel 2
 - OS: Android 11
 - Browser Chrome
 - Version 88.0.4324.181

**Additional context**
I tried to do some cleanup (disconnect, stopNotifications) in ngOnDestroy but i doesn't get called in livereload
stopNotifications before startNotifications results in Errors.
Is there any other way to do the cleanup when using livereload?

pentabarf, 2021-03-10:
I found a workaround.
One can use _window.addEventListener("beforeunload", ()=>{})_  to do the cleanup

pwespi, 2021-03-14:
Thanks for this issue and for sharing your workaround.

Which version of the plugin are you using?

dennisameling, 2021-03-17:
Can confirm that workaround works for live reload scenarios:

```TS
// If we're on hot reload, automatically disconnect on page refresh
window.addEventListener('beforeunload', () => {
  console.log('Browser "beforeunload" triggered. Disconnecting from device...');
  BleClient.disconnect(device.deviceId);
});
```

pentabarf, 2021-03-17:
sometimes 'beforeunload' takes to long and gets interrupted. for this case there is a little "hack" to stretch the time. you can do something like this:
`      let x = 500;
      let a = (new Date()).getTime() + x;`

pwespi, 2021-03-20:
There have been some improvements in the latest version. Could you check if the cleanup is still necessary in v0.5.0 (for Capacitor v2) and v1.0.0-2 (for Capacitor v3)?

pwespi, 2021-04-03:
I'm closing this with the assumption that the latest version has helped. Let me know if this is still an issue.

closed 2021-04-03 by pwespi

ZaLiTHkA, 2023-01-10:
so.. I've just discovered this same issue with the Android side of my current project, running with `@capacitor/core@4.5.0` and `@capacitor-community/bluetooth-le@2.1.0`.

@pwespi, before I log a new issue I just wanted to check: does this plugin have any way to "remove all listeners" that I could use at startup? I remember needing to do this with some other Cordova BLE plugins previously.