capacitor-community/bluetooth-le #69: Plugin with Ionic

aardrasystems, 2021-03-16:
Hi

I am very new to Capacitor plugins.

I am trying to install this plugin into my Ionic project so that I can use BLE on all platforms.

I have managed to install its as shown in the guide, but when I  do a build I am getting the following error messages:

Can you please advice what I am doing wrong?

`
ERROR in node_modules/@capacitor-community/bluetooth-le/dist/esm/definitions.d.ts:1:13 - error TS1005: '=' expected.

1 import type { PluginListenerHandle } from '@capacitor/core';
              ~
node_modules/@capacitor-community/bluetooth-le/dist/esm/definitions.d.ts:1:43 - error TS1005: ';' expected.

1 import type { PluginListenerHandle } from '@capacitor/core';
                                            ~~~~~~~~~~~~~~~~~
node_modules/@capacitor-community/bluetooth-le/dist/esm/web.d.ts:3:13 - error TS1005: '=' expected.

3 import type { BleDevice, BluetoothLePlugin, ConnectOptions, ReadOptions, ReadResult, RequestBleDeviceOptions, WriteOptions } from './definitions';
              ~
node_modules/@capacitor-community/bluetooth-le/dist/esm/web.d.ts:3:131 - error TS1005: ';' expected.

3 import type { BleDevice, BluetoothLePlugin, ConnectOptions, ReadOptions, ReadResult, RequestBleDeviceOptions, WriteOptions } from './definitions';
                                                                                                                                    ~~~~~~~~~~~~~~~
node_modules/@capacitor-community/bluetooth-le/dist/esm/bleClient.d.ts:1:13 - error TS1005: '=' expected.

1 import type { PluginListenerHandle } from '@capacitor/core';
              ~
node_modules/@capacitor-community/bluetooth-le/dist/esm/bleClient.d.ts:1:43 - error TS1005: ';' expected.

1 import type { PluginListenerHandle } from '@capacitor/core';
                                            ~~~~~~~~~~~~~~~~~
node_modules/@capacitor-community/bluetooth-le/dist/esm/bleClient.d.ts:2:13 - error TS1005: '=' expected.

2 import type { BleDevice, RequestBleDeviceOptions, ScanResult } from './definitions';
              ~
node_modules/@capacitor-community/bluetooth-le/dist/esm/bleClient.d.ts:2:69 - error TS1005: ';' expected.

2 import type { BleDevice, RequestBleDeviceOptions, ScanResult } from './definitions';
                                                                      ~~~~~~~~~~~~~~~

[ERROR] An error occurred while running subprocess ng.
`

pwespi, 2021-03-16:
Hi

Thanks for trying this plugin.

Can you post your `package.json`? I suspect that your app is using an old TypeScript version.

aardrasystems, 2021-03-16:
Thank You for quick reply:

This is the file:

{
  "name": "aardraapp",
  "version": "0.0.1",
  "author": "Ionic Framework",
  "homepage": "https://ionicframework.com/",
  "scripts": {
    "ng": "ng",
    "start": "ng serve",
    "build": "ng build",
    "test": "ng test",
    "lint": "ng lint",
    "e2e": "ng e2e"
  },
  "private": true,
  "dependencies": {
    "@angular-devkit/architect": "^0.901.0",
    "@angular/common": "^9.0.6",
    "@angular/core": "^9.0.6",
    "@angular/forms": "^9.0.6",
    "@angular/platform-browser": "^9.0.6",
    "@angular/platform-browser-dynamic": "^9.0.6",
    "@angular/router": "^9.0.6",
    "@capacitor-community/bluetooth-le": "^0.4.0",
    "@capacitor/android": "^2.4.7",
    "@capacitor/core": "^2.4.7",
    "@capacitor/ios": "^2.4.7",
    "@ionic-native/app-version": "^5.21.5",
    "@ionic-native/barcode-scanner": "^5.21.5",
    "@ionic-native/core": "^5.21.5",
    "@ionic-native/local-notifications": "^5.21.5",
    "@ionic-native/push": "^5.21.5",
    "@ionic-native/qr-scanner": "^5.21.5",
    "@ionic-native/splash-screen": "^5.21.5",
    "@ionic-native/sqlite": "^5.21.5",
    "@ionic-native/status-bar": "^5.21.5",
    "@ionic/angular": "^5.0.5",
    "build": "^0.1.4",
    "chart.js": "^2.9.3",
    "chartjs-plugin-zoom": "^0.7.5",
    "cordova": "^9.0.0",
    "cordova-android": "^8.1.0",
    "cordova-ios": "^5.1.0",
    "core-js": "^2.6.11",
    "echarts": "^4.6.0",
    "highcharts": "^7.2.1",
    "ionic": "^5.4.16",
    "ios": "0.0.1",
    "jetifier": "^1.6.5",
    "ngx-echarts": "^4.2.2",
    "rxjs": "~6.5.4",
    "tslib": "^1.10.0",
    "zone.js": "~0.10.2"
  },
  "devDependencies": {
    "@angular-devkit/build-angular": "~0.900.6",
    "@angular/cli": "9.0.6",
    "@angular/compiler": "^9.0.6",
    "@angular/compiler-cli": "^9.0.6",
    "@angular/language-service": "^9.0.6",
    "@capacitor/cli": "^2.4.7",
    "@ionic/angular-toolkit": "^2.2.0",
    "@types/jasmine": "~2.8.8",
    "@types/jasminewd2": "^2.0.8",
    "@types/node": "^12.11.1",
    "codelyzer": "^5.1.2",
    "jasmine-core": "~2.99.1",
    "jasmine-spec-reporter": "~4.2.1",
    "karma": "~3.1.4",
    "karma-chrome-launcher": "~2.2.0",
    "karma-coverage-istanbul-reporter": "^2.0.6",
    "karma-jasmine": "~1.1.2",
    "karma-jasmine-html-reporter": "^0.2.2",
    "protractor": "^5.4.3",
    "ts-node": "^8.0.3",
    "tslint": "~5.12.0",
    "typescript": "~3.7.5"
  },
  "description": "An Ionic project",
  "cordova": {
    "plugins": {},
    "platforms": [
      "android"
    ]
  }
}

aardrasystems, 2021-03-16:
I have also done a npm install -g typescript, but made no differnce

aardrasystems, 2021-03-16:
Hi pwespi

 

I have uploaded the file contents on github.

 

Thanks

 

DJ

aardrasystems, 2021-03-16:
I managed to get typescript 3.8.3 installed and the errors seems to have gone. I will now progress with the plugin and see if I can get it working.

pwespi, 2021-03-16:
Good that updating TypeScript helped.

closed 2021-03-16 by pwespi

aardrasystems, 2021-06-21:
Hi 

I am trying to run my project on a mac, 

In regards to TypeScript i am now getting 

ERROR in The Angular Compiler requires TypeScript >=3.6.4 and <3.8.0 but 3.8.3 was found instead.

What version of typescript it compatible with this plugin?

pwespi, 2021-07-03:
I have just tested using the plugin with TypeScript 3.7 and 3.8 and didn't have any issues. Generally I'm using TypeScript 4.x.

It seems like the problem is between TypeScript and Angular.

aardrasystems, 2021-07-05:
Hi

 

I update my Angular version, and it seems to solve the issue.

 

When I use the plugin on an Ios device, it seems like BLE is not recognized. Must I enable anything additional on Xcode or in my code?

 

 

Best Regards

 

Dharmesh Joshi 

 

 <http://www.aardra.co.uk> www.aardra.co.uk