capacitor-community/bluetooth-le #70: Notify not working after writing

swendleysprott, 2021-03-16:
**Describe the bug**
Notify not working after writing to BLE device on Android.

**Plugin version:**
- @capacitor-community/bluetooth-le: 0.4.0

**Smartphone (please complete the following information):**
- Device: OnePlus 8 
- OS: Android 11

**Logs from Android Studio**

D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
D/Device: resolve: setNotifications|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e Notification enabled.
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 23593037, pluginId: BluetoothLe, methodName: write
V/Capacitor: callback: 23593037, pluginId: BluetoothLe, methodName: write, methodData: {"deviceId":"E9:4C:F8:D1:E2:6C","service":"6e400001-b5a3-f393-e0A9-e50e24dcca9e","characteristic":"6e400002-b5a3-f393-e0A9-e50e24dcca9e","value":"fd 7b 22 43 61 65 73 63 22 3a 7b 22 69 76 22 3a 22 44 44 30 43 43 42 39 46 33 37 38 30 39 42 34 44 38 30 30 41 37 34 41 38 38 35 38 37 34 41 31 42 22 2c 22 6b 65 79 49 6e 64 65 78 22 3a 31 7d 7d fe"}
D/Device: resolve: write|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400002-b5a3-f393-e0a9-e50e24dcca9e Characteristic successfully written.
V/Capacitor/BluetoothLe: Notifying listeners for event notification|E9:4C:F8:D1:E2:6C|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e
D/Capacitor/BluetoothLe: No listeners found for event notification|E9:4C:F8:D1:E2:6C|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e

pwespi, 2021-03-16:
Thanks for using this plugin. Unfortunately, I cannot reproduce this issue. Could you show the code where you are doing the `startNotifications` and `write`? Are you using `BleClient` as mentioned in the [README](https://github.com/capacitor-community/bluetooth-le#usage)? Does it work on iOS or web?

swendleysprott, 2021-03-16:
Yes sure. i'll share it with you.

I imported the plugin as mentioned:

`import {BleClient, numbersToDataView, numberToUUID, textToDataView} from '@capacitor-community/bluetooth-le';`

starting notifications:

`
    async main(): Promise<void> {
        try {
            await BleClient.initialize();

          this.BleTestDevice = await BleClient.requestDevice(
              {
                name: 'AXASWODEMO'
              });

            await BleClient.connect(this.BleTestDevice.deviceId);

            console.log('connected to device', this.BleTestDevice);

            await BleClient.startEnabledNotifications(
                value => {
                    console.log(value)
                },
            );

            await BleClient.startNotifications(
                this.BleTestDevice.deviceId,
                TEST_SERVICE,
                TX_CHARACTERISTIC,
                value => {
                    this.parseHeartRate(value)
                }
            ).then(value => {
                console.log('Listening to notfications')
            });

        } catch (error) {
            console.error(error);
        }
    }`

For writing i use a button with the following function:

`    async write(): Promise<void> {
        try {
            await BleClient.initialize();

            const caesc = '{"Caesc":{"iv":"DD0CCB9F37809B4D800A74A885874A1B","keyIndex":1}}';
            await BleClient.write(
                this.BleTestDevice.deviceId,
                TEST_SERVICE,
                RX_CHARACTERISTIC,
                textToDataView(this.hex_to_ascii(START_BYTE) + caesc + this.hex_to_ascii(END_BYTE))
            );
            console.log('Caesc is geschreven naar device');
        } catch (e) {
            console.log(e)
        }
    }`

I havent tried it on web or ios.

pwespi, 2021-03-20:
Thanks for the details. There have been a few fixes n v0.5.0 (for Capacitor v2) and v1.0.0-2 (for Capacitor v3).

Could you try with the latest version?

swendleysprott, 2021-04-01:
Hi, I updated the version to v.0.5.1 but still experienced the same problem

pwespi, 2021-04-03:
Unfortunately I still cannot reproduce this. What's your @capacitor/core and @capacitor/android version? 

Could you add the logs from Android Studio when running this with v0.5.1?

swendleysprott, 2021-05-20:
Sorry for my late response im now running v0.6.0.

These are the versions i'm using.

`    "@capacitor/android": "^2.4.0",
    "@capacitor/core": "2.4.0"`

My logs in Androids studio are:

V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 31461040, pluginId: BluetoothLe, methodName: isBonded
V/Capacitor: callback: 31461040, pluginId: BluetoothLe, methodName: isBonded, methodData: {"deviceId":"F1:1D:BE:E3:8C:24"}
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 31461041, pluginId: BluetoothLe, methodName: connect
V/Capacitor: callback: 31461041, pluginId: BluetoothLe, methodName: connect, methodData: {"deviceId":"F1:1D:BE:E3:8C:24"}
D/BluetoothGatt: connect() - device: F1:1D:BE:E3:8C:24, auto: false
    registerApp()
    registerApp() - UUID=18c38ee6-e081-4d92-a2cf-bb9efaa2e14a
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=13
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=13 device=F1:1D:BE:E3:8C:24
D/Device: Connected to GATT server. Starting service discovery.
D/BluetoothGatt: discoverServices() - device: F1:1D:BE:E3:8C:24
D/BluetoothGatt: onSearchComplete() = Device=F1:1D:BE:E3:8C:24 Status=0
D/Device: requestMtu 512
D/BluetoothGatt: configureMTU() - device: F1:1D:BE:E3:8C:24 mtu: 512
D/BluetoothGatt: onConfigureMTU() - Device=F1:1D:BE:E3:8C:24 mtu=250 status=0
D/Device: MTU changed: 250
    resolve: connect Connected.
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 31461042, pluginId: BluetoothLe, methodName: addListener
V/Capacitor: callback: 31461042, pluginId: BluetoothLe, methodName: addListener, methodData: {"eventName":"notification|F1:1D:BE:E3:8C:24|6e400001-b5a3-f393-e0A9-e50e24dcca9e|6e400003-b5a3-f393-e0A9-e50e24dcca9e"}
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 31461043, pluginId: BluetoothLe, methodName: startNotifications
V/Capacitor: callback: 31461043, pluginId: BluetoothLe, methodName: startNotifications, methodData: {"deviceId":"F1:1D:BE:E3:8C:24","service":"6e400001-b5a3-f393-e0A9-e50e24dcca9e","characteristic":"6e400003-b5a3-f393-e0A9-e50e24dcca9e"}
D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
D/Device: resolve: setNotifications|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e Setting notification succeeded.
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 31461044, pluginId: BluetoothLe, methodName: write
V/Capacitor: callback: 31461044, pluginId: BluetoothLe, methodName: write, methodData: {"deviceId":"F1:1D:BE:E3:8C:24","service":"6e400001-b5a3-f393-e0A9-e50e24dcca9e","characteristic":"6e400002-b5a3-f393-e0A9-e50e24dcca9e","value":"fd 7b 22 43 61 65 73 63 22 3a 7b 22 69 76 22 3a 22 44 44 30 43 43 42 39 46 33 37 38 30 39 42 34 44 38 30 30 41 37 34 41 38 38 35 38 37 34 41 31 42 22 2c 22 6b 65 79 49 6e 64 65 78 22 3a 31 7d 7d fe"}
D/Device: resolve: write|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400002-b5a3-f393-e0a9-e50e24dcca9e Characteristic successfully written.
I/Capacitor/Console: File: http://192.168.2.5:8100/login-login-module.js - Line 1129 - Msg: Caesc is geschreven naar device
V/Capacitor/BluetoothLe: Notifying listeners for event notification|F1:1D:BE:E3:8C:24|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e
D/Capacitor/BluetoothLe: No listeners found for event notification|F1:1D:BE:E3:8C:24|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e

hecko, 2021-07-29:
@pwespi Im actually having an issue with this too - Im either not understanding the docs correctly or Im missing something. 

Im subscribing to the same Nordic NUS characteristic as @swendleysprott trying to listen for the data like this:

```
  async connectToDevice(deviceId) {
    console.log("Stop scanning first...")
    await this.BluetoothStopScanning();

    // this will check whether we are already connected or not and is we are, disconnect first
    await this.BluetoothDisconnect();

    console.log("Connecting to: " + deviceId);
    await BleClient.connect(deviceId);
    console.log("connected to device");
    this.setState({connectedToDeviceId: deviceId});

    await BleClient.startNotifications(deviceId, NUS_SERVICE_UUID, NUS_TX_CHARACTERISTIC, value => {
      console.log('received data from NUS: ' + value);
    });
    console.log("subscribed to characteristic");
  }
```

What Im seeing in logcat is this:

```
2021-07-30 00:15:26.566 20712-25608/com.example.app V/Capacitor/BluetoothLe: Notifying listeners for event notification|F6:5F:B4:EE:1A:B3|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e
2021-07-30 00:15:26.566 20712-25608/com.example.app D/Capacitor/BluetoothLe: No listeners found for event notification|F6:5F:B4:EE:1A:B3|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e
```

Which seems like im correctly subscribed to the charactersitic, but Im not seeing the data in console.log (via Chrome DevTools connected directly to Android 11)

So my question would be - is there a reason the subscribed data are not passed to the startNotifications() value handler? Am I understanding the docs correctly here?

majianjia, 2021-08-03:
@pwespi @hecko 
I am too having the same issue. 

~~~
2021-08-03 22:59:16.775 12580-12719/io.ionic.starter V/Capacitor: callback: 14554192, pluginId: BluetoothLe, methodName: startEnabledNotifications, methodData: {}
2021-08-03 22:59:16.779 12580-12580/io.ionic.starter I/Capacitor/Console: File: http://localhost/ - Line 213 - Msg: undefined
2021-08-03 22:59:16.779 12580-12580/io.ionic.starter I/Capacitor/Console: File: http://localhost/js/chunk-69312c9a.3b039f35.js - Line 3 - Msg: enable notification on:E8:EB:1B:53:D0:99
2021-08-03 22:59:16.780 12580-12719/io.ionic.starter V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 14554193, pluginId: BluetoothLe, methodName: addListener
2021-08-03 22:59:16.780 12580-12719/io.ionic.starter V/Capacitor: callback: 14554193, pluginId: BluetoothLe, methodName: addListener, methodData: {"eventName":"notification|E8:EB:1B:53:D0:99|49535343-FE7D-4AE5-8FA9-9FAFD205E455|49535343-1E4D-4BD9-BA61-23C647249616"}
2021-08-03 22:59:16.782 12580-12719/io.ionic.starter V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 14554194, pluginId: BluetoothLe, methodName: startNotifications
2021-08-03 22:59:16.782 12580-12719/io.ionic.starter V/Capacitor: callback: 14554194, pluginId: BluetoothLe, methodName: startNotifications, methodData: {"deviceId":"E8:EB:1B:53:D0:99","service":"49535343-FE7D-4AE5-8FA9-9FAFD205E455","characteristic":"49535343-1E4D-4BD9-BA61-23C647249616"}
2021-08-03 22:59:16.783 12580-12662/io.ionic.starter D/BluetoothGatt: setCharacteristicNotification() - uuid: 49535343-1e4d-4bd9-ba61-23c647249616 enable: true
2021-08-03 22:59:16.802 12580-12593/io.ionic.starter D/Device: resolve: setNotifications|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616 Setting notification succeeded.
2021-08-03 22:59:16.804 12580-12580/io.ionic.starter I/Capacitor/Console: File: http://localhost/ - Line 213 - Msg: undefined
2021-08-03 22:59:16.822 12580-12593/io.ionic.starter V/Capacitor/BluetoothLe: Notifying listeners for event notification|E8:EB:1B:53:D0:99|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616
2021-08-03 22:59:16.822 12580-12593/io.ionic.starter D/Capacitor/BluetoothLe: No listeners found for event notification|E8:EB:1B:53:D0:99|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616
2021-08-03 22:59:16.865 12580-12593/io.ionic.starter D/BluetoothGatt: onConnectionUpdated() - Device=E8:EB:1B:53:D0:99 interval=36 latency=0 timeout=500 status=0
2021-08-03 22:59:17.330 12580-12593/io.ionic.starter D/BluetoothGatt: onConnectionUpdated() - Device=E8:EB:1B:53:D0:99 interval=24 latency=0 timeout=512 status=0
2021-08-03 22:59:17.846 12580-12593/io.ionic.starter V/Capacitor/BluetoothLe: Notifying listeners for event notification|E8:EB:1B:53:D0:99|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616
2021-08-03 22:59:17.846 12580-12593/io.ionic.starter D/Capacitor/BluetoothLe: No listeners found for event notification|E8:EB:1B:53:D0:99|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616
2021-08-03 22:59:18.881 12580-12593/io.ionic.starter V/Capacitor/BluetoothLe: Notifying listeners for event notification|E8:EB:1B:53:D0:99|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616
2021-08-03 22:59:18.882 12580-12593/io.ionic.starter D/Capacitor/BluetoothLe: No listeners found for event notification|E8:EB:1B:53:D0:99|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616
2021-08-03 22:59:19.870 12580-12593/io.ionic.starter V/Capacitor/BluetoothLe: Notifying listeners for event notification|E8:EB:1B:53:D0:99|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616
2021-08-03 22:59:19.871 12580-12593/io.ionic.starter D/Capacitor/BluetoothLe: No listeners found for event notification|E8:EB:1B:53:D0:99|49535343-fe7d-4ae5-8fa9-9fafd205e455|49535343-1e4d-4bd9-ba61-23c647249616

~~~

majianjia, 2021-08-05:
@pwespi 
I think I found the issue. 
It is because I use the UPPER CASE for UUID. But the listener list key for the callback is lower case.

@Fllynn @hecko 
The current workaround is to use lower case only in UUID. 

[image]

Fllynn, 2021-08-05:
Well it has replaced the undefined with
[image]

```
2021-08-05 15:23:41.306 22984-27937/com.example.app V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 120255447, pluginId: BluetoothLe, methodName: addListener
2021-08-05 15:23:41.306 22984-27937/com.example.app V/Capacitor: callback: 120255447, pluginId: BluetoothLe, methodName: addListener, methodData: {"eventName":"notification|F4:4B:64:42:E7:6A|6E400001-B5A3-F393-E0A9-E50E24DCCA9E|6e400001-b5a3-f393-e0a9-e50e24dcca9e"}
2021-08-05 15:23:41.315 22984-27937/com.example.app V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 120255448, pluginId: BluetoothLe, methodName: startNotifications
2021-08-05 15:23:41.316 22984-27937/com.example.app V/Capacitor: callback: 120255448, pluginId: BluetoothLe, methodName: startNotifications, methodData: {"deviceId":"F4:4B:64:42:E7:6A","service":"6E400001-B5A3-F393-E0A9-E50E24DCCA9E","characteristic":"6e400001-b5a3-f393-e0a9-e50e24dcca9e"}
2021-08-05 15:23:41.320 22984-28093/com.example.app D/BluetoothGatt: onConnectionUpdated() - Device=F4:4B:64:42:E7:6A interval=36 latency=0 timeout=500 status=0
2021-08-05 15:23:41.322 22984-27858/com.example.app D/Device: reject: setNotifications|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400001-b5a3-f393-e0a9-e50e24dcca9e Characteristic not found.
2021-08-05 15:23:41.323 22984-27858/com.example.app D/Capacitor: Sending plugin error: {"save":false,"callbackId":"120255448","pluginId":"BluetoothLe","methodName":"startNotifications","success":false,"error":{"message":"Characteristic not found."}}
2021-08-05 15:23:41.337 22984-22984/com.example.app E/Capacitor/Console: File: http://localhost/ - Line 426 - Msg: Uncaught (in promise) Error: Characteristic not found.
2021-08-05 15:23:42.705 22984-27937/com.example.app V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 120255449, pluginId: BluetoothLe, methodName: removeListener
2021-08-05 15:23:42.705 22984-27937/com.example.app V/Capacitor: callback: 120255449, pluginId: BluetoothLe, methodName: removeListener, methodData: {"eventName":"onScanResult","callbackId":"120255444"}
2021-08-05 15:23:42.713 22984-27937/com.example.app V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 120255450, pluginId: BluetoothLe, methodName: stopLEScan
2021-08-05 15:23:42.714 22984-27937/com.example.app V/Capacitor: callback: 120255450, pluginId: BluetoothLe, methodName: stopLEScan, methodData: {}
```

majianjia, 2021-08-05:
@Fllynn The log shows your notification is rejected, I think you need to check whether your characteristic UUID is correct. I believe at least android support both upper and lower cases UUID.

They are different from your previous log.
New: 6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400001-b5a3-f393-e0a9-e50e24dcca9e
Old : 6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e
 
BTW, your service UUID is still in Capital.

Fllynn, 2021-08-05:
Even with the right UUID and lower cases same error 
```
2021-08-05 16:14:18.378 6099-9286/com.example.app V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 93262798, pluginId: BluetoothLe, methodName: addListener
2021-08-05 16:14:18.378 6099-9286/com.example.app V/Capacitor: callback: 93262798, pluginId: BluetoothLe, methodName: addListener, methodData: {"eventName":"notification|F4:4B:64:42:E7:6A|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400001-b5a3-f393-e0a9-e50e24dcca9e"}
2021-08-05 16:14:18.385 6099-9286/com.example.app V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 93262799, pluginId: BluetoothLe, methodName: startNotifications
2021-08-05 16:14:18.386 6099-9286/com.example.app V/Capacitor: callback: 93262799, pluginId: BluetoothLe, methodName: startNotifications, methodData: {"deviceId":"F4:4B:64:42:E7:6A","service":"6e400001-b5a3-f393-e0a9-e50e24dcca9e","characteristic":"6e400001-b5a3-f393-e0a9-e50e24dcca9e"}
2021-08-05 16:14:18.388 6099-9253/com.example.app D/Device: reject: setNotifications|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400001-b5a3-f393-e0a9-e50e24dcca9e Characteristic not found.
2021-08-05 16:14:18.389 6099-9926/com.example.app D/BluetoothGatt: onConnectionUpdated() - Device=F4:4B:64:42:E7:6A interval=60 latency=0 timeout=400 status=0
2021-08-05 16:14:18.389 6099-9253/com.example.app D/Capacitor: Sending plugin error: {"save":false,"callbackId":"93262799","pluginId":"BluetoothLe","methodName":"startNotifications","success":false,"error":{"message":"Characteristic not found."}}
2021-08-05 16:14:18.399 6099-6099/com.example.app E/Capacitor/Console: File: http://localhost/ - Line 426 - Msg: Uncaught (in promise) Error: Characteristic not found.
```

majianjia, 2021-08-05:
@Fllynn  no I meant you used the service UUID as the characteristic. which results in "characteristic not found"
6e400001-b5a3-f393-e0a9-e50e24dcca9e -> service
6e400002-b5a3-f393-e0a9-e50e24dcca9e -> TX characteristic
6e400003-b5a3-f393-e0a9-e50e24dcca9e -> RX characteristic

Fllynn, 2021-08-05:
Okay, so with the fixed UUIDs still the callback is `undefined`

```
2021-08-05 22:06:40.376 5268-9692/com.example.app V/Capacitor: callback: 114486288, pluginId: BluetoothLe, methodName: startNotifications, methodData: {"deviceId":"F4:4B:64:42:E7:6A","service":"6e400001-b5a3-f393-e0a9-e50e24dcca9e","characteristic":"6e400003-b5a3-f393-e0a9-e50e24dcca9e"}
2021-08-05 22:06:40.379 5268-9607/com.example.app D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
2021-08-05 22:06:40.387 5268-9943/com.example.app D/BluetoothGatt: onConnectionUpdated() - Device=F4:4B:64:42:E7:6A interval=36 latency=0 timeout=500 status=0
2021-08-05 22:06:40.433 5268-9943/com.example.app D/Device: resolve: setNotifications|6e400001-b5a3-f393-e0a9-e50e24dcca9e|6e400003-b5a3-f393-e0a9-e50e24dcca9e Setting notification succeeded.
2021-08-05 22:06:40.438 5268-5268/com.example.app I/Capacitor/Console: File: http://localhost/ - Line 213 - Msg: undefined
```

hecko, 2021-08-05:
@majianjia you suggestion to change UUIDs to lowercase seems to work around the issue in my case. Ill investigate a bit more and get back to this thread in a few hours. Good catch!

hecko, 2021-08-05:
@majianjia I can actually confirm this being the case. the UUIDs need to be lowercase for notifications to work properly. 

In my other Java code I actually have this: 

```
import java.util.UUID;

...

private static final UUID UART_TX_CHARACTERISTIC_UUID =
      UUID.fromString("6E400003-B5A3-F393-E0A9-E50E24DCCA9E");

...

mTXCharacteristic = service.getCharacteristic(UART_TX_CHARACTERISTIC_UUID);

...

          setNotificationCallback(mTXCharacteristic)
              .with(
                  (device, data) -> {
                    final String text = data.getStringValue(0);
                    log(LogContract.Log.Level.APPLICATION, "rx: \"" + text + "\"\n");
                    mCallbacks.onDataReceived(device, text);
                  });
          requestMtu(260).enqueue();
          enableNotifications(mTXCharacteristic).enqueue();

```

which works and might help improving this codes Kotlin implementation

pwespi, 2021-08-05:
Hi all

Thank you all for reporting and  debugging this issue. I can confirm that using uppercase UUIDs caused this behavior. I just released v1.1.3 which allows uppercase as well as lowercase UUIDs and should solve this.

@Fllynn In case v1.1.3 does not solve your problem, please open a new issue.

Thanks again.

PS: it was actually documented in the README that the plugin expects lowercase UUIDS, but it was easy to miss and the plugin should have handled this better.

closed 2021-08-05 by pwespi

majianjia, 2021-08-05:
@pwespi Yes indeed, it is on the front page of the document. Just saw that this afternoon. 
Great job on making this plugin, appreciated it.