capacitor-community/bluetooth-le #73: Cannot preform multiple pairing

aardrasystems, 2021-03-17:
I am not sure if I have found a bug or am I doing something wrong.

What I am trying to do is scan for my ble device, connect to it, start the pairing scheme(typing the in number) and once successful repeat the whole process again. The reason for repeating the process of pairing is incase my device was to fail for any given reason, a user would reset the device and start the pairing process again.

What I am experiencing is that once my device is paired, if I was to repeat the cycle again on my app, I am able to scan and see the device, but when I connect I do not get an option to retype the pin in this 2nd instance. But I was re repeat this process again, the pin input screen does appear. It seems that screen appear every other time.

This is my code 
async  ble_test(){
  console.log("Starting BLE TEST");      
  
  

  try {

      await BleClient.initialize();

    
    console.log('BLE INT');    
    await BleClient.requestLEScan(
      {
       // services: [HEART_RATE_SERVICE],
       name:"3400-DEMO",
      },
      result => {
        console.log('received new scan result', result);     
        this.Azure.ble_device_id = result.device.deviceId;
        this.connecttodevice(this.Azure.ble_device_id);
      },
    ); 

    setTimeout(async () => {
      await BleClient.stopLEScan();
      console.log('stopped scanning');
    }, 5000);
  } catch (error) {
    console.error(error);
  }
}

Thanks

pwespi, 2021-03-20:
Which platform are you using? iOS? If yes, on iOS there is no API to handle bonding. It is performed automatically when accessing a characteristic that requires security. See e.g. https://devzone.nordicsemi.com/f/nordic-q-a/8507/getting-an-ios-central-app-to-bond

aardrasystems, 2021-03-20:
No I am testing on Android. 
What I discovered, I need to manually unpair the device it I want re connect with pin screen. Is there anything in library to force a unpairing?

pwespi, 2021-04-03:
On Android, there is no public API for `removeBond`, see [BluetoothDevice](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#public-methods). So no, there is nothing in the library to force unbonding.

Also this plugin does not yet expose the `createBond` method on Android, see #50. So I don't even know how you managed to start the bonding process?

aardrasystems, 2021-04-03:
Hi

Maybe it could be my device, Once i start the pairing process, i get this pop up from the device requesting i enter the pin e..g 123456

pwespi, 2021-04-10:
`createBond` and `isBonded`have been added for Android.

closed 2021-04-10 by pwespi