capacitor-community/bluetooth-le #84: Scan for device -> connect to found device not working after first time (android)

squio, 2021-03-23:
**Describe the bug**

Scanning for an existing and available BLE device and then connecting works the very first time my app is installed and run but fails always on subsequent runs.

**To Reproduce**
In my code I basically have these steps:

- BleClient.initialize()
- BleClient.requestLEScan() -> scan for one device name and store the found device object
- BleClient.stopLEScan()
- BleClient.connect(deviceId) -> try to connect to the found device

This works the very first time the app is installed and run but fails always on subsequent runs.

I kept the default timeout for connecting (which is 30 seconds I believe) but the connection fails almost immediately.
The console shows:

`D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=8 device=B6:24:53:6D:8A:4D`

**Expected behavior**
The device is discovered (it is) and connected to (it isn't)

**work around**
I discovered a work-around: to disconnect before trying to connect

- BleClient.initialize()
- BleClient.requestLEScan() -> scan for one device name and store the found device object
- BleClient.stopLEScan()
- *BleClient.disconnect(deviceId);* -> this fails with error "device not connected"; ignore
- BleClient.connect(deviceId) -> now connect to the found device

This appears to solve the issue completely for my Android device

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.0.0-3

**Desktop (please complete the following information):**
- OS: MacOS 11
- Browser N/A
- Version Ionic 6.13.1 / Angular 9.1.14 / Capacitor 3.0.0-rc.0

**Smartphone (please complete the following information):**
- Device: Nokia 6.1
- OS: Android 10
- Browser System Webview (chrome)
- Version N/A

**Additional context**
Android studio console log output, the relevant part:

```
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 91203091, pluginId: BluetoothLe, methodName: addListener
V/Capacitor: callback: 91203091, pluginId: BluetoothLe, methodName: addListener, methodData: {"eventName":"disconnected|C9:54:C0:6D:30:4C"}
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 91203092, pluginId: BluetoothLe, methodName: connect
V/Capacitor: callback: 91203092, pluginId: BluetoothLe, methodName: connect, methodData: {"deviceId":"C9:54:C0:6D:30:4C"}
D/BluetoothGatt: connect() - device: C9:54:C0:6D:30:4C, auto: false
    registerApp()
D/BluetoothGatt: registerApp() - UUID=15cda7ec-0be6-47ba-be2a-65c5e240fb58
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
D/Device: reject: connect Connection timeout.
D/Capacitor: Sending plugin error: {"save":false,"callbackId":"91203092","pluginId":"BluetoothLe","methodName":"connect","success":false,"error":{"message":"Connection timeout."}}
I/Capacitor/Console: File: http://localhost/ ... - Msg: Connect - Connection failed: Error: Connection timeout.

D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=8 device=C9:54:C0:6D:30:4C
V/Capacitor/BluetoothLe: Notifying listeners for event disconnected|C9:54:C0:6D:30:4C
D/Device: Disconnected from GATT server.
W/Device: Resolve callback not registered for key: disconnect
I/Capacitor/Console: File: capacitor-runtime.js - Line 427 - Msg: undefined
I/Capacitor/Console: File: http://localhost/... - Msg: onDisconnect event callback from BLE, device ID: C9:54:C0:6D:30:4C
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 91203093, pluginId: BluetoothLe, methodName: stopNotifications
```

pwespi, 2021-04-03:
Thanks for using this plugin and for the detailed issue description.

Unfortunately, the error `status=133` you're getting from Android is kind of a generic error and doesn't say much about what is going wrong. According to [this article](https://punchthrough.com/android-ble-guide/) a possible workaround is to call `close()` on `BluetoothGatt`. This is what happens when you call `BleClient.disconnect(deviceId)`, so you have already found the appropriate workaround. I don't know what else could be done.

Could you check if the error also happens with another Android phone? I cannot reproduce it with my phones.

squio, 2021-04-06:
Thanks for your answer, the linked article is really helpful (and the Android Bluetooth stack really intimidating!).

I currently don't have another Android phone available regrettably but will try and see if someone else from my team is able to check this further, will let you know if we find anything new. And I'll leave the work around in place for now.

pwespi, 2021-04-10:
FYI in the newest version, `BleClient.disconnect`doesn't fail anymore when the device is not connected.

squio, 2021-04-12:
This looks like an excellent work around, I'll propose to document this #98 and close this issue?

closed 2021-04-20 by pwespi

subhamdandapat, 2021-08-17:
After successfully connecting not getting device peripheral object and for that unable to implement further process like startnotification or read data