capacitor-community/bluetooth-le #91: Reading the multiple characteristic with the same UUID from the same service

aardrasystems, 2021-04-05:
Hi,

I am not sure if this is a bug issue or something I am doing wrong.

I have a BLE device with WIFI module(WINC3400). 

I am able to perform a WIFI scan by writing to the corresponding Characteristics, but the results are read from particular Characteristics. It seems that each result is saved under the same Characteristics UUID.

When i perform a read on that Characteristics, it simply only outputting the first result only, which is the first Characteristics, but not the remaining ones.

So how can I read multiple Characteristics, which share the same UUID?

pwespi, 2021-04-06:
A characteristic can only have one value at a given time. But it can change its value over time.

I don't know your device, but it might be that it emits the results on one characteristic as notification one after the other. If that's the case, you should us [`startNotifications`](https://github.com/capacitor-community/bluetooth-le#startnotifications) (and `stopNotifications`) instead of `read`.

aardrasystems, 2021-04-06:
Hi, Yes make more sense as my device is scanning its finding new values.

When every I do a startNotifications, my APP just crashes and closes. This is on an Andriod.  Is this a known issue?

aardrasystems, 2021-04-07:
Should the characteristic I am reading from the need to support Notifications?

This the error we are getting when setup the notification:

`E/Capacitor: Serious error executing plugin
    java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Method.invoke(Native Method)
        at com.getcapacitor.PluginHandle.invoke(PluginHandle.java:99)
        at com.getcapacitor.Bridge$1.run(Bridge.java:542)
        at android.os.Handler.handleCallback(Handler.java:914)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:224)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'byte[] android.bluetooth.BluetoothGattDescriptor.getValue()' on a null object reference
        at android.bluetooth.BluetoothGatt.writeDescriptor(BluetoothGatt.java:1255)
        at com.capacitorjs.community.plugins.bluetoothle.Device.setNotifications(Device.kt:241)
        at com.capacitorjs.community.plugins.bluetoothle.BluetoothLe.startNotifications(BluetoothLe.kt:317)
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.getcapacitor.PluginHandle.invoke(PluginHandle.java:99) 
        at com.getcapacitor.Bridge$1.run(Bridge.java:542) 
        at android.os.Handler.handleCallback(Handler.java:914) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:224) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
E/AndroidRuntime: FATAL EXCEPTION: CapacitorPlugins
    Process: com.aardra.cloud, PID: 31543
    java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
        at com.getcapacitor.Bridge$1.run(Bridge.java:551)
        at android.os.Handler.handleCallback(Handler.java:914)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:224)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Method.invoke(Native Method)
        at com.getcapacitor.PluginHandle.invoke(PluginHandle.java:99)
        at com.getcapacitor.Bridge$1.run(Bridge.java:542)
        at android.os.Handler.handleCallback(Handler.java:914) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:224) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
     Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'byte[] android.bluetooth.BluetoothGattDescriptor.getValue()' on a null object reference
        at android.bluetooth.BluetoothGatt.writeDescriptor(BluetoothGatt.java:1255)
        at com.capacitorjs.community.plugins.bluetoothle.Device.setNotifications(Device.kt:241)
        at com.capacitorjs.community.plugins.bluetoothle.BluetoothLe.startNotifications(BluetoothLe.kt:317)
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.getcapacitor.PluginHandle.invoke(PluginHandle.java:99) 
        at com.getcapacitor.Bridge$1.run(Bridge.java:542) 
        at android.os.Handler.handleCallback(Handler.java:914) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:224) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
I/Process: Sending signal. PID: 31543 SIG: 9

I have attached the document of the BLE device for your reference:
[Atmel-42683-ATWINC3400-BLE-WiFi-Scan-and-Connect-Services-Guide_UserGuide.pdf](https://github.com/capacitor-community/bluetooth-le/files/6271032/Atmel-42683-ATWINC3400-BLE-WiFi-Scan-and-Connect-Services-Guide_UserGuide.pdf)
`

The service is fb8c0001-d224-11e4-85a1-0002a5d5c51b’ and characteristic is "AP Details"(fb8c0100-d224-11e4-85a1-0002a5d5c51b)

Thanks

pwespi, 2021-04-10:
> When every I do a startNotifications, my APP just crashes and closes. This is on an Andriod. Is this a known issue?

This was a bug in the plugin and has been fixed in the latest release. Thanks for reporting this.

pwespi, 2021-04-10:
> Should the characteristic I am reading from the need to support Notifications?

Yes, the characteristic has to support notifications, otherwise you cannot use `startNotifications`.

pwespi, 2021-04-10:
> The service is fb8c0001-d224-11e4-85a1-0002a5d5c51b’ and characteristic is "AP Details"(fb8c0100-d224-11e4-85a1-0002a5d5c51b)

> Is it also possible to do a READ but select a particular instance of characteristics to read? Based on what I have read it is possible to a service could have multiple characteristics with the same UUID. https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic#getInstanceId()

Each characteristic needs to have it's own UUID. There cannot be several characteristics behind the same UUID. (But several characteristics can be in the same service.) Where did you find this UUID `fb8c0100-d224-11e4-85a1-0002a5d5c51b`? I didn't find it in the document you linked. Maybe the second WiFi is just under `fb8c0101-d224-11e4-85a1-0002a5d5c51b` and so on? You can use the [nrf connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to inspect your device and find all characteristics.

aardrasystems, 2021-04-10:
That UUID is what is used by the WINC3400 module. It seems that their library assigns the same UUID for multiple characteristics. I have now managed to modify it so that each characteristic now has its own UUID. 

Therefore we can close this case.

closed 2021-04-10 by aardrasystems