capacitor-community/bluetooth-le #96: BleClient.requestLEScan gives exception 'TypeError: navigator.bluetooth.requestLEScan is not a function' in Chrome

hjvanranden, 2021-04-09:
**Describe the bug**
BleClient.requestLEScan gives exception 'TypeError: navigator.bluetooth.requestLEScan is not a function' in Chrome, but works fine on my Android tablet.
BleClient.requestDevice works fine in Chrome and on my Android tablet.

**To Reproduce in Chrome**
await BleClient.initialize();
await BleClient.requestLEScan({services: ['01ee0100-ba5e-f4ee-5ca1-eb1e5e4b1ce0']}, (scanResult: ScanResult) => {
          this.driversSvc.addDriver({device: scanResult.device});
        });

**Observed behavior**
exception 'TypeError: navigator.bluetooth.requestLEScan is not a function'

**Expected behavior**
No exception, yields found device as it does on my Android tablet.

**Plugin version etc.:**
{
  "name": "uvapp",
  "version": "0.0.1",
  "author": "Ionic Framework",
  "homepage": "https://ionicframework.com/",
  "scripts": {
    "ng": "ng",
    "start": "ng serve",
    "build": "ng build",
    "test": "ng test",
    "lint": "ng lint",
    "e2e": "ng e2e"
  },
  "private": true,
  "dependencies": {
    "@angular/common": "~11.2.0",
    "@angular/core": "~11.2.0",
    "@angular/forms": "~11.2.0",
    "@angular/platform-browser": "~11.2.0",
    "@angular/platform-browser-dynamic": "~11.2.0",
    "@angular/router": "~11.2.0",
    "@capacitor-community/bluetooth-le": "^0.5.1",
    "@capacitor/android": "^2.4.7",
    "@capacitor/core": "2.4.7",
    "@ionic/angular": "^5.5.2",
    "rxjs": "~6.6.0",
    "tslib": "^2.0.0",
    "zone.js": "~0.10.2"
  },
  "devDependencies": {
    "@angular-devkit/build-angular": "~0.1102.4",
    "@angular/cli": "~11.2.4",
    "@angular/compiler": "~11.2.0",
    "@angular/compiler-cli": "~11.2.0",
    "@angular/language-service": "~11.2.0",
    "@capacitor/cli": "2.4.7",
    "@ionic/angular-toolkit": "^3.1.1",
    "@types/jasmine": "~3.6.0",
    "@types/jasminewd2": "~2.0.3",
    "@types/node": "^12.11.1",
    "codelyzer": "^6.0.0",
    "jasmine-core": "~3.6.0",
    "jasmine-spec-reporter": "~5.0.0",
    "karma": "~5.2.0",
    "karma-chrome-launcher": "~3.1.0",
    "karma-coverage": "~2.0.3",
    "karma-coverage-istanbul-reporter": "~3.0.2",
    "karma-jasmine": "~4.0.0",
    "karma-jasmine-html-reporter": "^1.5.0",
    "protractor": "~7.0.0",
    "ts-node": "~8.3.0",
    "tslint": "~6.1.0",
    "typescript": "~4.0.2"
  },
  "description": "An Ionic project"
}

**Desktop (where it fails):**
- OS: Microsoft Windows 10 Enterprise, Version	10.0.17763 Build 17763
- Browser: Google Chrome Versie 89.0.4389.114 (Officiële build) (64-bits)

**Tablet (where it works fine):**
- Device: Samsung Galaxy Tab A (2016)
- OS: Android 8.1.0

hjvanranden, 2021-04-09:
This bug is relevant for me because testing in Chrome (ionic serve with hot reload) gives a much faster development cycle than testing on an Android tablet (ionic capacitor copy android, then run from Android studio).

pwespi, 2021-04-10:
`requestLEScan` is behind a flag in Chrome (`chrome://flags/#enable-experimental-web-platform-features`). See https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md#scanning-api.

Unfortunately, even if you enable this flag, there are still some bugs that make `requestLEScan`not very stable to use. E.g. on Windows there is this: https://bugs.chromium.org/p/chromium/issues/detail?id=1137504

However, you can use live-reloading with Capacitor and an Android device with `ionic capacitor run android -l --external`. See https://ionicframework.com/docs/cli/livereload.

pwespi, 2021-04-20:
I'm closing this as I don't think there is anything this plugin can do here. Let me know if you think otherwise.

closed 2021-04-20 by pwespi

Frtrillo, 2022-05-25:
> `requestLEScan` is behind a flag in Chrome (`chrome://flags/#enable-experimental-web-platform-features`). See [...]

works amazing the live reloader