capacitor-community/bluetooth-le #104: WriteWithoutResponse permitted to characteristic with response (Android)

squio, 2021-04-23:
**Describe the bug**
First up, I'm not sure this even is a bug or just a platform dependent difference which doesn't harm.

We have a custom device with a custom characteristic which *does* create a response and doesn't support the `withoutResponse` option.

in iOS this is clearly visible in the error log:

```text
[CoreBluetooth] WARNING:
Characteristic <CBCharacteristic: 0x283ced200, UUID = 65(...)66, properties = 0x8, value = (null), notifying = NO>
does not specify the "Write Without Response" property - ignoring response-less write
```

On Android the very same call sequence *does* work without any error messages.

**To Reproduce**
Steps to reproduce the behavior:
1. connect to bluetooth device
2. select a characteristic which implements write with response only
2. send data using `writeWithoutResponse()`
3. check for success 
4. check error messages (debugger log output in Xcode / Android Studio respectively)

**Expected behavior**
An error message stating that the `writeWithoutResponse` is not allowed

- iOS: error is shown as stated above
- Android: no error message, the `writeWithoutResponse()` call is successful

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.0.0-4

**Desktop (please complete the following information):**
- OS: Mac OS 11.2.3
- N/A
- N/A

**Smartphone (please complete the following information):**
- Device: iPhone 8 / Noklia 6.1
- OS: iOS 14.4.2 / Android 10
- Browser: System webview
- Version app built using Capacitor 3.0.0

pwespi, 2021-05-23:
Thanks a lot for reporting this.

However, I'm not sure there is anything that can be done about this in the plugin. I'm using the write type [WRITE_TYPE_NO_RESPONSE](https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic#WRITE_TYPE_NO_RESPONSE) in Android for `writeWithoutResponse`. I assume Android does something under the hood that when the characteristic does not support WRITE_TYPE_NO_RESPONSE, it falls back to [WRITE_TYPE_DEFAULT](https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic#WRITE_TYPE_DEFAULT). 

Since the Android Bluetooth API does not report an error, the plugin also does not report an error. So I would consider this a platform dependent difference which doesn't harm.

Or does this behavior cause a big problem for you?

squio, 2021-05-24:
Not a problem at all apart from being surprised by it: I started working on the Android version of an app which failed when compiling for iOS due to this difference. Changing the call to default `write()` solved it correctly for both platforms. Just something to be aware of I guess...

pwespi, 2021-06-05:
Thanks for the explanation.

As mentioned above I don't think there is anything that can be done in the plugin. So I'm closing this for issue for now.

closed 2021-06-05 by pwespi