capacitor-community/bluetooth-le #120: Writing text to characteristic fails

UltimateBrent, 2021-05-26:
**Describe the bug**
Using `.write()` doesn't seem to work for any DataView I send. I tried `textToDataView('string')` and `new DataView( new TextEncoder().encode('string').buffer )` but neither one seem to get written to the endpoint (bleno running on a raspberry pi). Using LightBlue, it works fine, and other Pis are able to write to it, so I don't think it's the receiver. The log in XCode says it was sent successfully, it just doesn't get there.

Reads also work fine on this characteristic, it's just the writes that have an issue.

```
[log] - - writing: log::hello-1622051334558
⚡️  To Native ->  BluetoothLe write 48144307
Resolve write|0000EC00-0000-1000-8000-00805F9B34FB|0000EC0E-0000-1000-8000-00805F9B34FB Successfully written value.
⚡️  TO JS undefined
```

**To Reproduce**
Raspberry Pi 3b running a Bleno server.
Performing the write from an iPad, if that matters.

**Expected behavior**
I expect the message to go through.

**Plugin version:**
- @capacitor-community/bluetooth-le: [e.g. 0.4.0]

**Desktop (please complete the following information):**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Smartphone (please complete the following information):**
- Device: Ipad Pro 2018
- OS: 14.4.2

pwespi, 2021-06-05:
Thanks for using this plugin.

Unfortunately, I cannot reproduce this and without an error from XCode, I don't know what to do. When you get `Successfully written value` after using `write`, this means that iOS responded successfully with [`didWriteValueFor`](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/1518823-peripheral).

Do you get any logs from you bleno server if a write was attempted at all? Could you try it with another device?

UltimateBrent, 2021-06-05:
Sorry it was for work so I couldn't fool with it much more and used a different plug-in. 

Hopefully it's isolated to whatever I was doing. I'll close it out.

closed 2021-06-05 by UltimateBrent