capacitor-community/bluetooth-le #121: App crashing when using Android TV

nik736, 2021-05-26:
Hello,

first of all I would like to say thanks for this library!

**Describe the bug**
I have developed an Android TV app, one part of it is scanning for BLE signals in the background. I am using different devices, for example a Fire TV stick, Mi TV Stick and some days ago I received the Nokia TV streaming box 8000 with which the app is crashing quite fast (after around 60 minutes). What the app does is run for 24h and just have the requestLEScan method running every 10 seconds.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the example code for the scanning API wrapped in a setInterval function that is running every 10 seconds.
2. After around 1 hour I am receiving the following error in the Android Studio console:

```
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 121276222, pluginId: BluetoothLe, methodName: stopLEScan
V/Capacitor: callback: 121276222, pluginId: BluetoothLe, methodName: stopLEScan, methodData: {}
D/DeviceScanner: Stop scanning.
D/BluetoothAdapter: isLeEnabled(): ON
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.app, PID: 31896
    java.util.ConcurrentModificationException
        at java.util.ArrayList$Itr.next(ArrayList.java:860)
        at com.getcapacitor.Plugin.notifyListeners(Plugin.java:671)
        at com.getcapacitor.Plugin.notifyListeners(Plugin.java:684)
        at com.capacitorjs.community.plugins.bluetoothle.BluetoothLe.access$notifyListeners(BluetoothLe.kt:37)
        at com.capacitorjs.community.plugins.bluetoothle.BluetoothLe$requestLEScan$2.invoke(BluetoothLe.kt:209)
        at com.capacitorjs.community.plugins.bluetoothle.BluetoothLe$requestLEScan$2.invoke(BluetoothLe.kt:37)
        at com.capacitorjs.community.plugins.bluetoothle.DeviceScanner$scanCallback$1.onScanResult(DeviceScanner.kt:71)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:486)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7356)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
I/Process: Sending signal. PID: 31896 SIG: 9
```
This only happens with the Nokia Box, Fire TV Stick and Mi Stick are working fine. Is there any way I can prevent it from crashing?

Any help is appreciated! Thanks so much.

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.0.0-4

- OS: Android
- Version 10, Stock Android TV running on the Nokia 8000 Box

pwespi, 2021-06-05:
Thanks for using this plugin.

Error handling has been improved in version 1.1.0. Could you check if this fixes your problem? (I cannot reproduce it and don't have access to a Nokia Box).

nik736, 2021-06-10:
Thanks for version 1.1.0. It's running fine for around 4 days now without crash, while still collecting BLE signals. I am not sure what the error was, since I can't find anything in the log, but trying to catch the error seem to have solved it.

Thanks!

closed 2021-06-10 by nik736