capacitor-community/bluetooth-le #126: isEnabled does not exist on type 'BleClientClass'

Raj-abm, 2021-06-08:
Hello sir,
I am using this bluetooth-le plugin in my project,
When I call the method isEnabled on BleClient, I am getting the error that, **property isEnabled does not exist on type 'BleClientClass'**. Please help me with this issue. I am working on Ionic react project and using capacitor version 3.
I followed your GitHub readme.md file.

Also let me know if this can scan mobile devices also, I am newbie recently joined a firm and working.
Thank you.

pwespi, 2021-06-08:
Thanks for using this plugin.

Please update the plugin to the latest version, currently 1.1.0. `isEnabled` was called `getEnabled` in previous versions (see [changelog](https://github.com/capacitor-community/bluetooth-le/blob/main/CHANGELOG.md)).

> Also let me know if this can scan mobile devices also, I am newbie recently joined a firm and working.

Yes, if mobile devices are advertising in Bluetooth peripheral mode, you can scan for them with this plugin

Raj-abm, 2021-06-08:
Thank you sir,
Thanks for your reply

closed 2021-06-10 by pwespi

udhayachandran03, 2024-03-14:
hi sir iam fascing an issue in blutoothle plugin iam working in the project of ionic angular, i want to connect and get the details of device using the bluetooth iam fascing error like this? Property 'getDeviceDetails' does not exist on type 'BleClientClass'. can you help me with this?