capacitor-community/bluetooth-le #127: Android app crashes when sending binary data to characteristic - StringIndexOutOfBoundsException

thankthemaker, 2021-06-09:
**Describe the bug**
Sending binary data to a characteristic does not work on Android. In Web-version it works as indended. It seems an empty value is send to the BLE stack

**Expected behavior**
Sending of binary data should be possible

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.1.0

**Smartphone (please complete the following information):**
- Device:HUWEI Mate 20 lite
- OS: Android 10

**Additional context**
Stacktrace:
```
2021-06-09 22:02:51.050 28610-28749/org.thankthemaker.rescueapp V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 45646988, pluginId: BluetoothLe, methodName: write
2021-06-09 22:02:51.051 28610-28749/org.thankthemaker.rescueapp V/Capacitor: callback: 45646988, pluginId: BluetoothLe, methodName: write, methodData: {"deviceId":"08:3A:F2:9A:E4:BE","service":"c8659210-af91-4ad3-a995-a58d6fd26145","characteristic":"c8659211-af91-4ad3-a995-a58d6fd26145","value":""}
2021-06-09 22:02:51.058 28610-28610/org.thankthemaker.rescueapp I/Capacitor/Console: File: http://localhost/ - Line 213 - Msg: undefined
2021-06-09 22:02:51.100 28610-28697/org.thankthemaker.rescueapp E/Capacitor: Serious error executing plugin
    java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Method.invoke(Native Method)
        at com.getcapacitor.PluginHandle.invoke(PluginHandle.java:121)
        at com.getcapacitor.Bridge.lambda$callPluginMethod$0$Bridge(Bridge.java:584)
        at com.getcapacitor.-$$Lambda$Bridge$25SFHybyAQk7zS27hTVXh2p8tmw.run(Unknown Source:8)
        at android.os.Handler.handleCallback(Handler.java:888)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:213)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.StringIndexOutOfBoundsException: length=0; index=0
        at java.lang.String.charAt(Native Method)
        at com.capacitorjs.community.plugins.bluetoothle.ConversionKt.hexToByte(Conversion.kt:23)
        at com.capacitorjs.community.plugins.bluetoothle.ConversionKt.stringToBytes(Conversion.kt:17)
        at com.capacitorjs.community.plugins.bluetoothle.Device.write(Device.kt:283)
        at com.capacitorjs.community.plugins.bluetoothle.BluetoothLe.write(BluetoothLe.kt:351)
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.getcapacitor.PluginHandle.invoke(PluginHandle.java:121) 
        at com.getcapacitor.Bridge.lambda$callPluginMethod$0$Bridge(Bridge.java:584) 
        at com.getcapacitor.-$$Lambda$Bridge$25SFHybyAQk7zS27hTVXh2p8tmw.run(Unknown Source:8) 
        at android.os.Handler.handleCallback(Handler.java:888) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:213) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
2021-06-09 22:02:51.101 28610-28697/org.thankthemaker.rescueapp I/QarthLog: [PatchStore] createDisableExceptionQarthFile
2021-06-09 22:02:51.101 28610-28697/org.thankthemaker.rescueapp I/QarthLog: [PatchStore] create disable file for org.thankthemaker.rescueapp uid is 10204
    
    --------- beginning of crash
2021-06-09 22:02:51.110 28610-28697/org.thankthemaker.rescueapp E/AndroidRuntime: FATAL EXCEPTION: CapacitorPlugins
    Process: org.thankthemaker.rescueapp, PID: 28610
    java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
        at com.getcapacitor.Bridge.lambda$callPluginMethod$0$Bridge(Bridge.java:593)
        at com.getcapacitor.-$$Lambda$Bridge$25SFHybyAQk7zS27hTVXh2p8tmw.run(Unknown Source:8)
        at android.os.Handler.handleCallback(Handler.java:888)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:213)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Method.invoke(Native Method)
        at com.getcapacitor.PluginHandle.invoke(PluginHandle.java:121)
        at com.getcapacitor.Bridge.lambda$callPluginMethod$0$Bridge(Bridge.java:584)
        at com.getcapacitor.-$$Lambda$Bridge$25SFHybyAQk7zS27hTVXh2p8tmw.run(Unknown Source:8) 
        at android.os.Handler.handleCallback(Handler.java:888) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:213) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
     Caused by: java.lang.StringIndexOutOfBoundsException: length=0; index=0
        at java.lang.String.charAt(Native Method)
        at com.capacitorjs.community.plugins.bluetoothle.ConversionKt.hexToByte(Conversion.kt:23)
        at com.capacitorjs.community.plugins.bluetoothle.ConversionKt.stringToBytes(Conversion.kt:17)
        at com.capacitorjs.community.plugins.bluetoothle.Device.write(Device.kt:283)
        at com.capacitorjs.community.plugins.bluetoothle.BluetoothLe.write(BluetoothLe.kt:351)
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.getcapacitor.PluginHandle.invoke(PluginHandle.java:121) 
        at com.getcapacitor.Bridge.lambda$callPluginMethod$0$Bridge(Bridge.java:584) 
        at com.getcapacitor.-$$Lambda$Bridge$25SFHybyAQk7zS27hTVXh2p8tmw.run(Unknown Source:8) 
        at android.os.Handler.handleCallback(Handler.java:888) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:213) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
2021-06-09 22:02:51.171 28610-28697/org.thankthemaker.rescueapp I/Process: Sending signal. PID: 28610 SIG: 9
```

thankthemaker, 2021-06-09:
Here's the code that is used to send the data and that works in Google Chrome on Desktop (macOS Big Sur)
```

    getFirmwareFile(version: string, ) : Observable<any> {
      return this.http.get(`${this.releaseUrl}/firmware-${version}.bin`, {
        responseType: 'arraybuffer'
      }).pipe(
        timeout(5000)
      )
    }

###SNIP###
    this.firmwareService.getFirmwareFile(this.version).subscribe(result => {
      this.totalSize = result.byteLength
      this.downloadFinished = true
      this.updateData = result
      this.loading.dismiss()
    })

###SNIP###

      this.dataToSend = this.updateData.slice(this.currentPosition, this.currentPosition + this.amountToWrite);

###SNIP###

        try {
          await BleClient.write(
            this.deviceId,
            AppSettings.OTA_SERVICE_UUID,
            AppSettings.FILE_CHAR_UUID,
            this.dataToSend
          )
        } catch(error) {
          console.error('Error BLE.write ' + error)
        }
```
Where dataToSend is an ArrayBuffer.

thankthemaker, 2021-06-10:
Ok, debugged and finally found a solution. The ArrayBuffer has to be wrapped ...

**Does not** work:
```
BleClient.write(
            this.deviceId,
            AppSettings.OTA_SERVICE_UUID,
            AppSettings.FILE_CHAR_UUID,
            this.dataToSend
          )
```

**Does** work

```
BleClient.write(
            this.deviceId,
            AppSettings.OTA_SERVICE_UUID,
            AppSettings.FILE_CHAR_UUID,
            new DataView(this.dataToSend)
          )
```

I'm wondering that the wrapping isn't needed in Chrome on macOS, but at least it now works.

closed 2021-06-10 by thankthemaker

pwespi, 2021-06-10:
Thanks for using this plugin and for the detailed issue. 

Good that you already found a solution. The `write` method expects indeed a `DataView` and not an `ArrayBuffer`. I'm surprised that it worked in Chrome without wrapping it.

Anyway it should not crash the app completely in Android, but just report an error. I'll take a look.

pwespi, 2021-06-11:
I just release v1.1.1 which will reject the `write` call if an `ArrayBuffer` is passed instead of a `DataView` on all platforms. The app should not crash anymore on Android, but just return an error. With a DataView everything should work as expected.

Thanks for reporting this.