capacitor-community/bluetooth-le #134: Error TS2665: Invalid module name in augmentation.

aardrasystems, 2021-06-21:
Hi 
I am trying to get plugin to compile on my Mac Ionic Project version, but i am getting following error message:

[ng] ERROR in node_modules/@capacitor-community/bluetooth-le/dist/esm/config.d.ts:1:16 - error TS2665: Invalid module name in augmentation. Module '@capacitor/cli' resolves to an untyped module at '/Users/user922263/Desktop/app/node_modules/@capacitor/cli/dist/index.js', which cannot be augmented.

Has anyone had this issue before?

pwespi, 2021-07-03:
I have never seen this problem before. Can you create a minimal repository that reproduces this problem?

aardrasystems, 2021-07-13:
HI

I updated the Angular version, and it seems to have solved the issue.

closed 2021-07-13 by aardrasystems