capacitor-community/bluetooth-le #141: Bluetooth connection not updating on Bluetooth LE device?

Raj-abm, 2021-07-05:
Hello sir,
I am using your plugin with Ionic React,
I followed your steps in Readme file,
I scan for BLE devices and when I click on particular device, the connection gets established but the connection status is not updating in the BLE device that is connection status is not updated to connected.
I thought this is the issue with BLE device itself, but it is working fine with another app that is written in Qt.
please do help me with this,

I am using bluetooth-le api 1.1.1, capacitor 3,
I am developing this for android

Raj-abm, 2021-07-06:
Please anybody react to this issue...

Raj-abm, 2021-07-15:
Also, I am able to connect even though the pairing is turned off on the ble device

pwespi, 2021-07-19:
What do you mean by "the connection status is not updating"? 

Connecting and pairing/bonding are not the same. So depending on the BLE device, you can still connect when pairing/bonding is turned off.

Raj-abm, 2021-07-20:
> What do you mean by "the connection status is not updating"?
> 
> Connecting and pairing/bonding are not the same. So depending on the BLE device, you can still connect when pairing/bonding is turned off.

"The connection status is not updating" means, when the connection is successful, I expect a Bluetooth icon to change i.e for example from gray to blue on the ble device, and
I also tested this on One plus ear buds, here the buds didn't blink on successful connection.

Regarding pairing/bonding - my ble device is designed such that, connection shouldn't happen when Bluetooth is on, but pairing is off, but it is being ignored by this plugin.

please help

Raj-abm, 2021-09-02:
> Has your problem been solved?
> 
> If not, can you provide more information, such as e.g. error logs, a reproduction repository etc.

I try again with the latest plugin and reply you.

Thank you

closed 2021-10-03 by github-actions[bot]