capacitor-community/bluetooth-le #142: BLE not appearing on IOS app version

aardrasystems, 2021-07-06:
Hi

I have followed all the steps shown in setting the plugin for IOS, but when I start my IOS app version there is no BLE request or check my apps "Access List" I do not see the Bluetooth there.

It works well on Andriod.

What could i be doing wrong?

Raj-abm, 2021-07-07:
Hi,
I need your help a bit,
I am using this plugin with android, you said it is working fine for you on android,
In my case I am able to make a successful connection with BLE device but the connection status is not updating in BLE device i.e the status should update from 'not connected' to 'connected', which is not happening.
This is the link to my issue [https://github.com/capacitor-community/bluetooth-le/issues/141](url)
please let me know what is the case with you,
also if you can share me that code part.
Thank you,

aardrasystems, 2021-07-07:
Hi  your link has nothing on it, can you please create a new post with your issue

Raj-abm, 2021-07-07:
> Hi your link has nothing on it, can you please create a new post with your issue

Hai, I have mentioned my issue in the above comment itself,
also you can reach my issue with this url 
https://github.com/capacitor-community/bluetooth-le/issues/142
you may need to copy paste it in url bar.

aardrasystems, 2021-07-07:
I am using Angular and not React, so i might not be the right person for this?

Has anyone got it working on Ios?

I am using capacitor 2.4.7, with  plugin version 0.6.0.

It seems that it can not preform the initial scan successfully.

aardrasystems, 2021-07-09:
I have updated my capacitor to version 3 and am i am using the latest version of the plugin.
Andriod works fine, but when i do a scan in IOS it fails to find the device and times out after my preset duration.

This is my code:

```
`  async  connect_to_BLE_Device(){
    this.ssid_array=[];
    this.message_display="1";
     console.log("Starting BLE TEST");         
     
    try {
        await BleClient.initialize();
        this.message_display="2";
        console.log('BLE INT');    
        await BleClient.requestLEScan(
        {        
         name:"3400-DEMO",
         allowDuplicates:false,
        },
        result => {
          console.log('received new scan result', result);     
          this.message_display="5";
          this.Azure.ble_device_id = result.device.deviceId;
          this.connect_to_deviceid()
        },
      );       
       setTimeout(async () => {
         await BleClient.stopLEScan();     
         this.message_display="3";
         console.log('stopped scanning')          
         this.connect_to_deviceid();

       }, 15000);
    } catch (error) {
      console.error(error);
      console.error("async connect_to_BLE_Device()");    
      this.message_display="4->"+error;

    }
  }
`
```

aardrasystems, 2021-07-13:
Any idea, what could be wrong?

pwespi, 2021-07-19:
Have you tried without the `name` filter? Can you find the device when you scan with another app on the same phone?

closed 2021-08-22 by github-actions[bot]