capacitor-community/bluetooth-le #149: Connection Timeout error on IoS

aardrasystems, 2021-07-18:
Hi 

When I try to connect to my BLE device I am getting a Connection Timeout Error. This is only noticeable on a IOS , I am using X code with my M1 chipset base Macbook to simulate the app.  After few times, I am able to connect to the device, but I am getting timeout issues very often. 

What would be the cause of this? Is there any information I can provide to help solve this issue?

pwespi, 2021-07-19:
It's hard to say what's going on here. 

Is the device close enough? Does it work on Android? Can you connect with another app to the device?

aardrasystems, 2021-07-22:
I think I might have solved the issue, but need to perform some more tests.

I noticed that on IOS I need to let it finish scanning before I try to connect to my device. Will on Andriod this did not seem to be a problem.

pwespi, 2021-08-21:
Has your issue been solved?

closed 2021-09-21 by github-actions[bot]

nemoneph, 2022-12-10:
same issue for me, @hristo0624 did you find a solution ?