capacitor-community/bluetooth-le #151: Check if device is connected

psande, 2021-07-19:
Hi!

I see there is a method for Android to check if the device is connected. Is there an equivalent for iOS? If not, how do I check if the device is already connected or if I lost connection with the device?

Thanks!

pwespi, 2021-07-19:
Currently the plugin doesn't expose a method to check if a device is connected. However, you can add a `onDisconnect` callback to `connect` that will be invoked when the connection is lost. See https://github.com/capacitor-community/bluetooth-le#connect

psande, 2021-07-19:
Hi @pwespi, thanks for your quick reply. I missed that event, I will give that a try otherwise I suppose I can just make a .read command to check if I'm still connected.

Thanks again!

closed 2021-07-19 by psande