capacitor-community/bluetooth-le #154: Supports for Indication

majianjia, 2021-08-04:
**Is your feature request related to a problem? Please describe.**
Indication support is also needed in some BLE modules which prefers to not use the notification in their generic transparent UART service. 
The one I use is RN4871, it seems always to change the Notification back to Indication. Which might result in the issues discussed here:
https://github.com/capacitor-community/bluetooth-le/issues/70#issuecomment-889500079
I guess, this is because the `enableNotification()` only register a callback for Notification but not Indication, results in:
` D/Capacitor/BluetoothLe: No listeners found for event notification|E8:EB:1B:53:D0:99`

**Describe the solution you'd like**
Add enable/disable Indication just like enableNotification() API

**Describe alternatives you've considered**
Allow the current notification callback to handle indication incoming message is also very handy. 

**Additional context**
Add any other context or screenshots about the feature request here.

majianjia, 2021-08-04:
I just made a test with some device that actually can send Notification. But the problem mentioned in https://github.com/capacitor-community/bluetooth-le/issues/70#issuecomment-889500079 still exist. 
It might not help to solve the issue.

pwespi, 2021-08-30:
The current method `startNotifications` already supports notifications as well as indications. 

On [iOS](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518949-setnotifyvalue) and [web](https://developer.mozilla.org/en-US/docs/Web/API/BluetoothRemoteGATTCharacteristic/startNotifications), there is actually no API to handle indications separately from notifications. Therefore I don't think there is a way to add a `startIndications` method to this plugin that would be supported on all platforms.

If something doesn't work for you, please open a new issue with steps on how to reproduce.

Thanks for using this plugin.

closed 2021-08-30 by pwespi