capacitor-community/bluetooth-le #157: pair method

sultanmyrza, 2021-08-08:
I'm trying to connect to GoPro https://gopro.github.io/OpenGoPro/tutorials/python/connect-ble
I'm following their tutorial they use python library called [Bleak](https://github.com/hbldh/bleak) to interact with GoPro

Bleak version is
```
client = BleakClient(bluetoothDevice) // GoPro in this case
await client.connect(timeout=15) // connects
```
The GoPro has encryption-protected characteristics which require us to pair before writing to them. Therefore now that we are connected, we need to attempt to pair:
```
try:
    await client.pair()
except NotImplementedError:
    # This is expected on Mac
    pass
```

## ionic version
I was able to connect to go pro with bluetooth but when I try to send command to GoPro form ionic app I got this
``` JavaScript
    const commandReqUUID = 'b5f90072-aa8d-11e3-9046-0002a5d5c51b';
    const service = '0000fea6-0000-1000-8000-00805f9b34fb';
    try {
      await BleClient.write(
        device.deviceId,
        service,
        commandReqUUID,
        numbersToDataView([3, 1, 1, 1])
      );
    } catch (error) {
      console.log(`error: ${JSON.stringify(error)}`);

    }
```
```
{"errorMessage":"Encryption is insufficient.","message":"Encryption is insufficient."}
```

## question 
will we have something like `BleClient.pair()` alternative to `BleacClient.connect(device).pair()`

pwespi, 2021-08-09:
Which platform are you using? If Android, there is the [`createBond`](https://github.com/capacitor-community/bluetooth-le#createbond) method (which is only available on Android). On iOS it should automatically trigger the bonding process when using an encrypted characteristic (see e.g. https://devzone.nordicsemi.com/f/nordic-q-a/8507/getting-an-ios-central-app-to-bond).

Another option (although less user friendly) is to use the Bluetooth settings of the OS to establish the bond.

sultanmyrza, 2021-08-09:
Right now I'm building for iOS but next step will be also add Android as well.
I will document my findings here maybe for other who do GoPro connection with ionic could benefit

pwespi, 2021-08-21:
The link was just to show that on iOS there is nothing to be done from the app code. Bonding is handled by the OS.

Did you manage to make it work?

sultanmyrza, 2021-08-22:
Hello yes got it working both on iOS and android

closed 2021-08-22 by sultanmyrza