capacitor-community/bluetooth-le #158: Combination with background-geolocation: No disconnect when app gets closed

pentabarf, 2021-08-09:
**Description**
Using this Plugin together with capacitor-community/background-geolocation in Android: bluetooth connection doesn't get closed when app is closed.

**Reproduce**
1. Create an ionic app for Android
2. install background-geolocation
3. install cordova-plugin-ble-central or capacitor-community/bluetooth-le
4. connect a ble device with the app
5. close the app
6. device is still connected with phone and therefore doesn't get find when reopen the app
7. to verify: disable Bluetooth in phone settings after app is closed and the device will notify a disconnect

**Expected behavior**
Bluetooth connection should get closed when app gets closed

**Smartphone**
 - Device: Motorola moto g 5G plus, Google Pixel 2
 - OS: Android 10
 - Version 29

**Additional context**
I am working on a ble-sensor combined with geolocation tracker project

I opened this also as an issue at capacitor-community/background-geolocation:
[https://github.com/capacitor-community/background-geolocation/issues/32](https://github.com/capacitor-community/background-geolocation/issues/32)

pwespi, 2021-08-21:
Unfortunately I wasn't able to reproduce your behavior.

However, in the newest release 1.2.0 there is a new method [getConnectedDevices](https://github.com/capacitor-community/bluetooth-le#getconnecteddevices) which might be useful for your situation. With this method you should be able to get the connected device without the need for scanning.

Does this solve your problem?

pentabarf, 2021-09-01:
Thanks for the advice with getConnectedDevices. After restarting the app I was able to get a list containing the device but i am not able to disconnect from it.

pwespi, 2021-09-06:
Did you connect to the device from the app? If you connect from another app or the OS settings, the app is not allowed to disconnect?

pentabarf, 2021-09-10:
I connect from an earlier instance of the app. As i wrote: the device doesn't get disconnect when killing the app whyever because of background-geolocation.

HyperLife1119, 2024-10-31:
> Thanks for the advice with getConnectedDevices. After restarting the app I was able to get a list containing the device but i am not able to disconnect from it.

I have the same problem, it can be reproduced on android 12, but not on android 14.