capacitor-community/bluetooth-le #170: getting read timeout on read characteristic from connected device (used in Ionic app)

sdetweil, 2021-08-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
the reads were started in parallel 
log of results, after scan, have array of peripherals
```
  [log] - peripheral info={"uuids":["00010202-27b9-42f0-82aa-2e951747bbf9"],"txPower":127,"localName":"FacilityDevice","device":{"name":"FacilityDevice","deviceId":"FF85CCC9-5925-DFB4-C2DE-A614F966D654"},"rssi":-63}
⚡️  [log] - peripheral info={"uuids":["00020202-27b9-42f0-82aa-2e951747bbf9"],"localName":"FacilityDevice","rssi":-69,"txPower":127,"device":{"deviceId":"9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD","name":"FacilityDevice"}}
⚡️  To Native ->  BluetoothLe connect 32443200
Connecting to peripheral <CBPeripheral: 0x283d592c0, identifier = FF85CCC9-5925-DFB4-C2DE-A614F966D654, name = FacilityDevice, mtu = 0, state = disconnected>
Connected to device <CBPeripheral: 0x283d592c0, identifier = FF85CCC9-5925-DFB4-C2DE-A614F966D654, name = FacilityDevice, mtu = 23, state = connected>
Resolve connect|FF85CCC9-5925-DFB4-C2DE-A614F966D654 Successfully connected.
Connected to peripheral. Waiting for service discovery.
didDiscoverServices
didDiscoverCharacteristicsFor 1 1
Resolve connect Connection successful.
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe connect 32443201
Connecting to peripheral <CBPeripheral: 0x283d595e0, identifier = 9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD, name = FacilityDevice, mtu = 0, state = disconnected>
Connected to device <CBPeripheral: 0x283d595e0, identifier = 9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD, name = FacilityDevice, mtu = 23, state = connected>
Resolve connect|9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD Successfully connected.
Connected to peripheral. Waiting for service discovery.
didDiscoverServices
didDiscoverCharacteristicsFor 1 1
Resolve connect Connection successful.
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe read 32443202
Reading value
Resolve callback not registered for key:  read|Optional("00010202-27B9-42F0-82AA-2E951747BBF9")|00009AFD-0000-1000-8000-00805F9B34FB
Reject read|00010202-27B9-42F0-82AA-2E951747BBF9|00009AFD-0000-1000-8000-00805F9B34FB Read timeout.
ERROR MESSAGE:  {"errorMessage":"Read timeout.","message":"Read timeout."}
⚡️  [error] - {"errorMessage":"Read timeout.","message":"Read timeout."}
⚡️  To Native ->  BluetoothLe read 32443203
Reading value
⚡️  [log] - BleClient read rssi error={"errorMessage":"Read timeout."}
Resolve callback not registered for key:  read|Optional("00010202-27B9-42F0-82AA-2E951747BBF9")|00009AFF-0000-1000-8000-00805F9B34FB
Reject read|00010202-27B9-42F0-82AA-2E951747BBF9|00009AFF-0000-1000-8000-00805F9B34FB Read timeout.
ERROR MESSAGE:  {"message":"Read timeout.","errorMessage":"Read timeout."}
⚡️  To Native ->  BluetoothLe read 32443204
⚡️  [error] - {"message":"Read timeout.","errorMessage":"Read timeout."}
Reading value
⚡️  [log] - BleClient read table error={"errorMessage":"Read timeout."}
Resolve callback not registered for key:  read|Optional("00020202-27B9-42F0-82AA-2E951747BBF9")|00009AFD-0000-1000-8000-00805F9B34FB
⚡️  [error] - [Cordova-Plugin-IBeacon] WARNING did not receive delegate ready callback from DOM after 30.000000 seconds!
Reject read|00020202-27B9-42F0-82AA-2E951747BBF9|00009AFD-0000-1000-8000-00805F9B34FB Read timeout.
ERROR MESSAGE:  {"errorMessage":"Read timeout.","message":"Read timeout."}
⚡️  To Native ->  BluetoothLe read 32443205
⚡️  [error] - {"errorMessage":"Read timeout.","message":"Read timeout."}
Reading value
⚡️  [log] - BleClient read rssi error={"errorMessage":"Read timeout."}
Resolve callback not registered for key:  read|Optional("00020202-27B9-42F0-82AA-2E951747BBF9")|00009AFF-0000-1000-8000-00805F9B34FB
Reject read|00020202-27B9-42F0-82AA-2E951747BBF9|00009AFF-0000-1000-8000-00805F9B34FB Read timeout.
ERROR MESSAGE:  {"errorMessage":"Read timeout.","message":"Read timeout."}
⚡️  [error] - {"errorMessage":"Read timeout.","message":"Read timeout."}
⚡️  [log] - BleClient read table error={"errorMessage":"Read timeout."}
```

**Expected behavior**
expect read to succeed

**Screenshots**

**Plugin version:**
- @capacitor-community/bluetooth-le: [e.g. 0.4.0]
- 1.2.0

**Desktop (please complete the following information):**
- OS: [e.g. iOS] mac osx big Sur, xcode 13

**Smartphone (please complete the following information):**
- Device: [e.g. iPhone6] iphone 7
- OS: [e.g. iOS8.1]   14.7

**Additional context**
Add any other context about the problem here.

the device says that the characteristics were read twice 
```
09:34:42.957 -> Connected event, central: 41:c2:1b:19:ad:9f
09:34:42.957 -> central rssi=-42
09:34:42.957 ->  address=41:c2:1b:19:ad:9f
09:34:44.475 -> RSSI characteristic read, central rssi=-42
09:34:44.475 -> RSSI characteristic read, central rssi=-42
09:34:49.824 -> hash characteristic read, central rssi=-42
09:34:49.824 -> hash characteristic read, central rssi=-42
09:35:28.576 -> Disconnected event, central: 41:c2:1b:19:ad:9f
```

code 
```
foundDevices.forEach((peripheral)=>{
               console.log("peripheral info="+JSON.stringify(peripheral))
               BleClient.connect(peripheral.device.deviceId).then(()=>{
                promises.push(new Promise((resolve, reject)=>{
                  BleClient.read(peripheral.device.deviceId, peripheral.uuids[0], numberToUUID("9AFD")).then((receivedRssi)=>{
                    console.log('RSSI value=', receivedRssi.getUint8(0));
                    resolve()
                  }).catch((error)=>{
                      console.log("BleClient read rssi error="+JSON.stringify(error))
                      reject(error)
                  })
                }))                  
                promises.push(new Promise((resolve, reject)=>{
                  BleClient.read(peripheral.device.deviceId, peripheral.uuids[0], numberToUUID("9AFF")).then((receivedTable)=>{
                    console.log('Table value=', receivedTable);
                    resolve()
                  }).catch((error)=>{
                      console.log("BleClient read table error="+JSON.stringify(error))
                      reject(error)
                  })       
                 }))
                 Promise.all(promises).then(()=>{
                     console.log("BleClient disconnecting")
                   BleClient.disconnect(peripheral.device.deviceId)
                 }).catch(error=>{
                    console.log("read chars error="+JSON.stringiy(error))
                    console.log("BleClient disconnecting")
                    BleClient.disconnect(peripheral.device.deviceId)
                 })
               }).catch((error)=>{
                  BleClient.disconnect(peripheral.device.deviceId)
                      console.log("BleClient connect error="+JSON.stringify(error))
                  })
             })
```

sdetweil, 2021-08-23:
here is a screen shot from Ble Scanner (on Android) for one of these devices 
[image]

pwespi, 2021-08-23:
Thanks for using this plugin.

Unfortunately, I cannot reproduce the behavior you're describing. However I tried to implement a possible fix based on your logs.

Could you try if your problem persist with the development build version `1.2.0-1-gb3062ec`?
```
npm i @capacitor-community/bluetooth-le@1.2.0-1-gb3062ec
```

sdetweil, 2021-08-23:
get compiler error in Xcode.. should have mentioned the CBService error before 
but the default  is new

also, was going to mention.. my arduino device WAS advertising in lowercase (that is what is received from the scan)
if I upper case it, it STILL appears as lower case... 

[image]

sdetweil, 2021-08-23:
if I fix both those errors like this 

I get a function not found on the uppercase 
[image]

sdetweil, 2021-08-24:
I had to do a clean and then build, but now it works..