capacitor-community/bluetooth-le #178: Turn on Bluetooth on device

LgArcida, 2021-08-31:
To automate the process and give a better experience to the end user.
Is it possible to turn on Bluetooth when invoking `Initialize` or as a separate method just like cordova's solution  [BLE-central](https://github.com/don/cordova-plugin-ble-central#enable)

mutasimissa, 2021-09-06:
@pwespi is it possible to make it on both Android and iOS? if yes I can work on the Android part

pwespi, 2021-09-06:
@mutasimissa No, as far as I know it's not possible to enable Bluetooth from an app on iOS. So this would be Android only. But if you want to have a go at it, I'm happy to accept a PR. I would put it in a separate method though and not change the existing initialize method.

LgArcida, 2021-09-06:
Should the current initialize method turn  device's Bluetooth on?

pwespi, 2021-09-06:
No, the current initialize method only handles permissions and checks if a Bluetooth adapter is available (on Android) or initializes the `CBCentralManager` (on iOS). It should not turn on Bluetooth as this is not available on iOS. 

After `initialize` you can use `isEnabled` on both Android and iOS to check whether Bluetooth is enabled or not, but there is not yet a method to enable it.

LgArcida, 2021-09-06:
Got it, is it in the roadmap to have a method to turn on BL on android devices at least ?

pwespi, 2021-09-06:
Yes, it's on the roadmap.

@mutasimissa Let me know if you can work on it, otherwise I'll take it.

LgArcida, 2021-09-06:
Thanks! Looking forward for it

mutasimissa, 2021-09-06:
Yes, I will work on it on a separate method @pwespi

pwespi, 2021-10-02:
I added `enable` and `disable` in the latest version.

(@mutasimissa I was adding some features in Android so I went ahead and implemented this. Hope you don't mind.)

pwespi, 2021-10-02:
closed by 53f1b3db51a4cc00547307b46e6dbde50c895d28

closed 2021-10-02 by pwespi