capacitor-community/bluetooth-le #179: Check location on Android

kozakhs, 2021-09-01:
Hello!
It would be very useful if this plugin is able to check if location is enabled or not on Android.
Permissions are handled in initialize() which si perfect, but the state of location not.

Thank you !!!

kozakhs, 2021-09-02:
> But can't you check the location permission of Geolocation via Capacitor?

Of course I can. But just due to location checking I need to install another plugin. This BT plugins have function isLocationEnabled():
https://github.com/randdusing/cordova-plugin-bluetoothle#islocationenabled
https://github.com/don/cordova-plugin-ble-central#islocationenabled 
... which is very handy for me...

Thank you for the answer.

mutasimissa, 2021-09-06:
I don't think you need to install another plugin, there is a method in Capacitor core to do that for you, see here and tell us the results:
https://capacitorjs.com/docs/v2/apis/permissions

kozakhs, 2021-09-16:
> I don't think you need to install another plugin, there is a method in Capacitor core to do that for you, see here and tell us the results:
> https://capacitorjs.com/docs/v2/apis/permissions

Sorry for my delayed response. Well it can be a solution, but what I am missing is the example, how to use this Permissions API .
There is also an opened issue (feature) on github: https://github.com/ionic-team/capacitor-site/issues/123

To save my time,  I will just install Geolocation plugin for now (how you mentioned in the first reply).  I will just hope that somebody of plugin maintainers will notice this enhancement and implement it in the future :-)

Thank you

squio, 2021-10-01:
The Permissions API /  plugin is only available for capacitor 2.

Also it is not clear if this really checks for the location services being enabled, or only whether the app has the reqeuested permission. Both are needed for BLE to work.

I suggested a feature for the geolocation plugin but it was declined: https://github.com/ionic-team/capacitor-plugins/issues/640

I'm willing to work on a PR, feedback is appreciated!

Copied from the linked issue:

> Description
> 
> Currently the checkPermissions() and requestPermissions() only check for and enable the App permissions.
> The device user can still have location services disabled - which is undetectable by this plugin.
> 
> On Android, location services are required to be enabled to do a Bluetooth scan request.
> If location services are not available, the BLE stack does not generate any error messages but just
> times out and discovers nothing (by design).
> 
> I want to be able to check whether location services have been disabled globally without trying to get an actual location because I actually don't want to request the user location at all.
> Platform(s)
> 
> - Android (primarily)
> -  ~iOS (could be useful as well?)~
> 
> Preferred Solution
> 
> Implement the solution from PR #589 as a callable plugin method, something along the lines of:
> 
> ```java
> public bool checkLocationServices()
> {
>     LocationManager lm = (LocationManager) context.getSystemService(Context.LOCATION_SERVICE);
>     return (LocationManagerCompat.isLocationEnabled(lm));
> }
> ```
> 
> A similar requestLocatuinServices() method - which requests the user to enable location services - could be helpful as well.
> Alternatives
> 
> As my use case is purely related to Bluetooth scanning, this could also be added in
> plugin @capacitor-community/bluetooth-le
> but the feature feels more at place in the actual geolocation plugin.
> Additional Context
> 
> It seems that similar functionality may exist in some (outdated?) cordova plugins but I'm very
> hesitant to pollute my project with duplicate and potentially outdated functionality.
> Alternative suggestions are welcome though!
> 
> 
>

pwespi, 2021-10-02:
Thank you all for the discussion and the suggestions. I added `isLocationEnabled` and `openLocationSettings` in the latest version. Hope it will be useful.

pwespi, 2021-10-02:
closed by 4c1cc60b76736c0693d7c7bfab1e093710289c68

closed 2021-10-02 by pwespi