capacitor-community/bluetooth-le #180: Not able to connect till allowDuplicates became true

aardrasystems, 2021-09-02:
Hi All

I am not sure if this is a bug or something I have done wrong.

My BLE device was able to communicate with my app on all my platform devices such as the Andriod phone, IPAD, Macbook Pro M1.

Until recently, after an update on my Macbook, I noticed my BLE device was no longer able to connect or communicate.

After some investigation I set 
`allowDuplicates: true`

After which i was able to connect to that device.

Now when i set
`allowDuplicates: false`

I am still able to connect to that device.

Is there a reason why this has happened?

Maybe it's best to keep it true.

mutasimissa, 2021-09-06:
I think the problem is that some devices send two different times, one with only device id and the other with full discovery information, so when you allow duplicates the device you connect to has the information that it needs in order to read/write characteristics.
Can you console.log the result of the scanning with and without allowing duplicates, filter out the device you are connecting to, I expect to see one with all information and the other with no information but only the deviceId

pwespi, 2021-09-06:
Hm it's hard to say what was going on without additional information.

> device was no longer able to connect or communicate

Did you get an error or what happened when you tried to connect?

aardrasystems, 2021-09-06:
Very unusually , as i am able to now see the device and connect to it.  I have tried numerous things, but can not replicate the issue.

There could be a possibility that it could be do with something on MacBook Pro.

As it working now , we can close this case.

closed 2021-09-06 by aardrasystems