capacitor-community/bluetooth-le #183: Check if device is connected

kozakhs, 2021-09-06:
Hello,

please, is it possible to implement function to check whether bluetooth device is connected?
For example:
isConnected(deviceId: string) => Promise<boolean> ?

Thank you very much!

mutasimissa, 2021-09-06:
I think you can use the method getconnectedDevices like this:

 const isConnected = await BleClient.getConnectedDevices([SOME_UUID_TO_FILTER_OR_GET_ALL])

kozakhs, 2021-09-06:
> I think you can use the method getconnectedDevices like this:
> 
> const isConnected = await BleClient.getConnectedDevices([SOME_UUID_TO_FILTER_OR_GET_ALL])

Thats exactly what I need. Thank you very much! I did not realise I can do it this way :)

mutasimissa, 2021-09-06:
You are welcome, Can you please close the issue?

closed 2021-09-06 by kozakhs