capacitor-community/bluetooth-le #189: AAPT: error: resource android:attr/lStar not found.

sultanmyrza, 2021-09-15:
**Describe the bug**
A clear and concise description of what the bug is.
```
Execution failed for task ':app:processDebugResources'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > Android resource linking failed
     /Users/sultanmyrza/.gradle/caches/transforms-2/files-2.1/462cf05f81b7c169f5ede52434de8415/core-1.7.0-alpha02/res/values/values.xml:105:5-114:25: AAPT: error: resource android:attr/lStar not found.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
```

**To Reproduce**
Steps to reproduce the behavior:
1.  create ionic project with capacitor 2
2. install bluetooth-le plugin `npm install @capacitor-community/bluetooth-le@0.6.0`
3. try to run project for android 
4. See error

**Expected behavior**
Start app on android

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Plugin version:**
-  "@capacitor-community/bluetooth-le": "^0.6.0",

sultanmyrza, 2021-09-15:
I found temporary solution for my case (android)

**Solution for Ionic:**

Simply search globally for `core-ktx` dependencies for all the ionic plugins and set the `1.6.0` version instead of `+`.

**From:**

    implementation "androidx.core:core-ktx:+"

**To:**
 
    implementation "androidx.core:core-ktx:1.6.0"

I alose did [pull request](https://github.com/capacitor-community/bluetooth-le/pull/188) hope can be approved

pwespi, 2021-09-21:
Thanks a lot. Closed by #194.

closed 2021-09-21 by pwespi