capacitor-community/bluetooth-le #192: Plugin does not work after disable/enable BT if device is connected

kozakhs, 2021-09-16:
**Describe the bug**
If I connect to a device (without bonding) on mobile phone and then I disable/enable bluetooth, the plugin stops work and do not react on any command.

**To Reproduce**
For example, you can reproduce the issue this way:
1)
```
await BleClient.connect('00:07:80:39:DF:F5', () => {
  // onDisconnect ...
});

console.log('Connected to the device');

setInterval(async () => {
            console.log('getting connected devices');
            const devices = await BleClient.getConnectedDevices([]);
            console.log('Connected devices: ', devices);
        }, 3000);
```
You will see in console every 3 second 2 messages above 
(the function must not be getConnectedDevices, it can be also for example BleClient.isEnabled() or whatever else)

2) Disable bluetooth on mobile phone and after few seconds enable bluetooth

3) You will see in console just 'getting connected devices' without second message  - the Promise will never be executed

**Expected behavior**
After bluetooth is enabled, everything works like before

**Plugin version:**
@capacitor-community/bluetooth-le  1.3.0

**Desktop (please complete the following information):**
Windows 10

**Smartphone (please complete the following information):**
Android 11, Samsung Galaxy A40
Android 9, Nokia TA-1032

kozakhs, 2021-09-29:
Please, any response?

pwespi, 2021-10-02:
Thank you for using this plugin. Unfortunately I cannot reproduce the bug you're describing (I've tried with Android 8 and 10). 

Do get any error messages in Android Studio?

kozakhs, 2021-10-18:
I am sorry for delayed reply. No errors are in Android Studio. However I have tried to make a clean capacitor project and also did not managed to reproduce the issue. It is strange that in one project it causes this behavior and in new clean project everything is alright. So probably something is wrong in my project. For this moment I am closing the issue and if I encounter this error again, I will open new one. Thank you for this great plugin and for your time.