capacitor-community/bluetooth-le #196: Support iOS 15 SDK - Build error

dmoulinier, 2021-09-22:
The build failed  with iOS 15 SDK. Due to changes in the CoreBluetooth API, The changes are related to nullability:

#### iOS 14.X
```swift
 CBService
 unowned(unsafe) var peripheral: CBPeripheral { get }
 
CBCharacteristic
 unowned(unsafe) var service: CBService { get }
 
 CBDescriptor
 unowned(unsafe) var characteristic: CBCharacteristic { get }
 ```
 
 #### iOS 15.X
 ```swift
 CBService
 weak var peripheral: CBPeripheral? { get }
 
 CBCharacteristic
 weak var service: CBService? { get }
 
 CBDescriptor
 weak var characteristic: CBCharacteristic? { get }
 ```

The main affected class is: Device.swift
 ```swift
 private func getKey(_ prefix: String, _ characteristic: CBCharacteristic) -> String {
        let serviceUUIDString = cbuuidToStringUppercase(characteristic.service.uuid)
        let characteristicUUIDString = cbuuidToStringUppercase(characteristic.uuid)
        return "\(prefix)|\(serviceUUIDString)|\(characteristicUUIDString)"
    }
 ```

With the build error:
```
../node_modules/@capacitor-community/bluetooth-le/ios/Plugin/Device.swift:187:72: Value of optional type 'CBService?' must be unwrapped to refer to member 'uuid' of wrapped base type 'CBService'
```

pwespi, 2021-09-22:
closed by #197

pwespi, 2021-09-22:
Thanks  for reporting this and finding the affected class.

closed 2021-09-22 by pwespi