capacitor-community/bluetooth-le #211: Connection status?

Raj-abm, 2021-10-05:
Hello thanks for the plugin.

Is there a way to check the connection status?
for examples, when the ble server disconnects or when bluetooth on the server is turned off, I should be able to know the current status and take action accordingly.

Thanks for reply.

squio, 2021-10-05:
Did you mean the `onDisconnect` callback when connecting?

Works perfectly well for me (iOS and Android).

Code fragment:

```ts
await BleClient.connect(
	device.deviceId,
	// onDisconnect:
	deviceId => {
		this.onBleDisconnected(deviceId);
	}
);
```

Documented here: [connect](https://github.com/capacitor-community/bluetooth-le#connect)

Raj-abm, 2021-10-05:
Thank you sir, it did solve.

closed 2021-10-05 by Raj-abm