capacitor-community/bluetooth-le #228: Android 12 support

Mr-Mime, 2021-11-02:
With the release of Android 12 [new permission](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions) got introduced, which affect the usage of BLE.

The location permissions are no longer needed to scan for BLE devices, which is a huge improvement, as the location permission seemed strange to some users.

Is this already supported by this plugin or are there plans to introduce it soon?
Asking as I might migrate to this plugin in near future.

pwespi, 2021-11-14:
Thank you for your interest in this plugin. 

The Android 12 permissions are not yet supported by this plugin. But it is planned to support them soon (either by adding them to this plugin or by allowing the app to set them).

pwespi, 2022-01-16:
Thank you for your patience. I started working on the Android 12 permissions and would appreciate it if someone could test it before I make a release.

To test it:

```bash
npm i @capacitor-community/bluetooth-le@1.6.1-2-g6ad3879
npx cap sync android
```
and follow the steps here: https://github.com/capacitor-community/bluetooth-le/tree/android-permissions#optional-android-12-bluetooth-permissions

Mr-Mime, 2022-01-18:
I haven't migrated my app to this plugin.
Also working on something different now. Don't think I'll be able to test soon :(

egeozer, 2022-02-10:
Hi, I’ve tested the custom version of @pwespi, done all the requirements in readme file for the optional permission feature, however I have gradle build errors whenever I attempt to build the app. This does not occur on the official release (1.7) version of this plugin.

pwespi, 2022-02-11:
@egeozer Thanks a lot for testing. Could you share the gradle build errors you're getting?

egeozer, 2022-02-11:
I finally managed to fix them! There were 2 issues preventing me from having a successful build.

1) In androidManifest file, make sure to add 'android:exported="<true or false>"' under <activity> tag.

2)Make sure to run on JDK 11+ as it is a requirement for angular 12 apparently.

Now that I have the app built, I will confirm next week if it still asks for location permissions on Android 12.

Thanks, Ege.

egeozer, 2022-02-15:
Update: On Android 12, with @pwespi's fix, the ble scanner manages to scan with location services turned off. However, not all bluetooth devices could be scanned (only 3 out of 10 got scanned). On Android 11 or IOS, the scanner manages to find all devices. What could be the reason of this?

Thanks!

Ege

pwespi, 2022-02-20:
@egeozer Thank you for your testing and feedback. I will add this to the readme before making a release.

About the missing devices, these were probably beacons. The Android documentation says:
> Note: If you include neverForLocation in your android:usesPermissionFlags, some BLE beacons are filtered from the scan results.
https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#assert-never-for-location

I will also add this to the readme.

closed 2022-02-27 by pwespi

pwespi, 2022-02-27:
Android 12 permissions support has been added in v1.8.0.

ZaLiTHkA, 2022-12-09:
I ran into a problem while updating my project's `AndroidManifest.xml` file, where the error message stated:

> The prefix "tools" for attribute "tools:targetApi" associated with an element type "uses-permissions" is not bound.

I did sort it out by adding `xmlns:tools="http://schemas.android.com/tools"` to my project's manifest root element, which I only discovered by looking at this library's own `AndroidManifest.xml` file.. could somebody perhaps update the `README.md` file to mention this, in case somebody else ends up in the same boat?

RodrigoRomero2308, 2022-12-22:
Same as @ZaLiTHkA, thanks for the comment

PR #274 (2022-02-27): feat(android): add support for Android 12 permissions
closes #228