capacitor-community/bluetooth-le #233: Autoconnect for already known devices

thankthemaker, 2021-11-07:
Is it possible to implement autoconnect for a device that was searched via requestDevice before? 
I'd really like to avoid scanning for my device every time I start my app, it should connect automatically.

Is this already possible with this plugin?

pwespi, 2021-11-14:
Yes, it should be possible. If you save the deviceId that you get from requestDevice in your first app run, you can then use [getDevices](https://github.com/capacitor-community/bluetooth-le#getdevices) with that deviceId to retrieve the device in subsequent app runs. If you're only using Android, you don't even have to use getDevices and can directly connect.

Is this what you were looking for?

thankthemaker, 2021-11-15:
Yes, that's what I was looking for. I'll give it a try. Thank you.

pwespi, 2021-11-15:
In case something does not work, let me know, and I will reopen this issue.

closed 2021-11-15 by pwespi