capacitor-community/bluetooth-le #234: Callback on notification termination from device

mutasimissa, 2021-11-07:
**Is your feature request related to a problem? Please describe.**
It's not possible to know if the notification stream is still active from a device or not

**Describe the solution you'd like**
Passing a callback function to be invoked when notifications from device is terminated

**Describe alternatives you've considered**
a separate function to be called to determine if there is an active notification or not

**Additional context**
N/A

pwespi, 2021-11-14:
Thanks you for this feature request. However I'm not sure if it's possible to implement. Do you know if the API for this even exists in Android, iOS or web? I'm not aware of any.

mutasimissa, 2021-11-15:
Yes, I understand there isn't an API to do that straightforward but is there any other way to know that the notifications are terminated, not receiving anymore or stopped?

mutasimissa, 2021-11-15:
If not, then I think that the as you stated in the #224 , the disconnect callback should be enough

pwespi, 2021-11-15:
Yes, I think the disconnect callback should be used. Or what's your use-case for a device that stops sending notifications but is still connected?

One other option could be to read the characteristic descriptor (requires plugin version >=1.6.0). 
```ts
const HEART_RATE_SERVICE = "0000180d-0000-1000-8000-00805f9b34fb";
const HEART_RATE_CHARACTERISTIC = "00002a37-0000-1000-8000-00805f9b34fb";
const CLIENT_CHARACTERISTIC_CONFIG = "00002902-0000-1000-8000-00805f9b34fb";

const value = await BleClient.readDescriptor(
  deviceId,
  HEART_RATE_SERVICE,
  HEART_RATE_CHARACTERISTIC,
  CLIENT_CHARACTERISTIC_CONFIG
);

value.getInt8(0) // 0 when notifications disabled, 1 when notifications enabled, 2 when indications enabled
```
But this probably does not work on iOS and polling it might also not be a good idea. I would suggest to just use the disconnect callback.

mutasimissa, 2021-11-17:
I tested on a couple of different devices and yes, when it stops notifications it's because the device is disconnected, thus the disconnect callback should be enough.
Thanks

closed 2021-11-17 by mutasimissa