capacitor-community/bluetooth-le #235: Manufacturer data on scan is empty

sinkeIng, 2021-11-09:
**Describe the bug**
When scanning with requestLEScan(...) result have empty Manufacturer data. As we can see from documentation, [scan result](https://github.com/capacitor-community/bluetooth-le/blob/0.x/README.md#scanresult) callback for manufacturer data is { [key: string]: DataView; } and in this plug is { [key: string]: String; }.
Android -> https://github.com/capacitor-community/bluetooth-le/blob/0.x/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/BluetoothLe.kt#L455
iOS -> https://github.com/capacitor-community/bluetooth-le/blob/0.x/ios/Plugin/Plugin.swift#L313

**To Reproduce**
For both platform just call requestLEScan(...) and log the result, manufacturer data will always be "key": "".

**Expected behavior**
Manufacturer data should be not empty if there is any.

**Plugin version:**
- @capacitor-community/bluetooth-le: 0.6.1 (we working with this version but this is issue for new version as well)

**Android**
- Device: Google Pixel 5a
- OS: Android 12

**iOS**
- Device: iPhone 6
- OS: 12.5.4

**Additional context**
If we check BluetoothLe.kt from Android here is passed string instead of DataView
https://github.com/capacitor-community/bluetooth-le/blob/main/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/BluetoothLe.kt#L678 `manufacturerData.put(key.toString(), bytesToString(bytes))`

**Fix**
For Android we need to create DataView object and send that instead of String. So fix for Android will be something like:
`val dataView = JSObject()`
`dataView.put("buffer", bytesToString(bytes))`
`dataView.put("byteLength", bytes.count())`
`dataView.put("byteOffset", 0)`
`manufacturerData.put(key.toString(), dataView)`

I will create PR with fix today.

pwespi, 2021-11-14:
Thank you for using this plugin.

I cannot reproduce the behavior you're describing. In my tests I receive the manufacturerData as expected.

Are you using the `BleClient` (and not the BluetoothLe class) as described in the [README](https://github.com/capacitor-community/bluetooth-le#usage)?

It is correct that the data is passed as string from the native layer to the webview, but it is then transformed into a DataView by the BleClient. The Capacitor bridge can only handle strings for now.

sinkeIng, 2021-11-16:
Hello @pwespi,
Thanks for fast answer and for plugin in general :)

Yes we are using BleClient as described in the README.

When I call scan with this:
`BleClient.requestLEScan({},
        (result) => {
          console.log(`scan: ${JSON.stringify(result)}`)
        },
      );`

Manufacturer data for all devices will look like this (key will be different of course) :
`"manufacturerData":{"196":{}},`

But if I add change described above I will get manufacturer data like this:
`"manufacturerData":{"196":{"buffer":"0E 32 7D 12 80 ","byteLength":5,"byteOffset":0}},`

That is log from Android phone. For iOS is the same issue, but actually there is one print from CAPBridge which prints manufacturer data populated (line 481) and console.log from scan will print empty manufacturer data.

If you have any more questions please don't hesitate to ask. Thanks in advance!

Milan

pwespi, 2021-11-16:
Thank you for the further details. This helps.

The problem is that a `DataView` is not JSON serializable, e.g. `JSON.stringify(new DataView(Uint8Array.from([1,2,3]).buffer))` will return `"{}"`.

You can try something like
```ts
BleClient.requestLEScan({}, (result) => {
  for (const key of Object.keys(result.manufacturerData ?? {})) {
    console.log(
      'manufacturerData',
      key,
      result.manufacturerData[key] ? dataViewToNumbers(result.manufacturerData[key]) : undefined
    );
  }
});
```

sinkeIng, 2021-11-17:
Thanks again for help and plugin :)

closed 2021-11-17 by sinkeIng