capacitor-community/bluetooth-le #247: startNotifications method not resolving

Raj-abm, 2021-11-18:
Hello sir,
Thank you for the plugin,
I am using the plugin version 1.5.0 in my Ionic React(Capacitor) app.
I am developing the app on windows 10 computer and running the app on android device version 10 and 11.

I am able to connect to Bluetooth low energy server and able read and write to a characteristic and also subscribe to notifications.
Everything works fine.

But sometimes `startNotifications` method does not resolve or reject (it is like, app pauses at start notifications method) and no further execution takes place.
please help,
Thank you

pwespi, 2021-11-19:
Thanks for using this plugin. 

It is hard to say what is going on. Do you get any errors in Android Studio? Can you share the logs from Android Studio? Does it happen with a specific characteristic, or on any characteristic? Can you continue using the app or does it freeze completely?

mutasimissa, 2021-11-20:
@Raj-abm can you please also try console.log something on the connect callback? something like
```
      await BleClient.connect(SOME_DEVICE_ID, (id) =>
        console.log(`Device ${id} disconnected!`)
      );
```

closed 2021-12-20 by github-actions[bot]