capacitor-community/bluetooth-le #249: Support for createBond() in iOS

aanders77, 2021-11-20:
Hi! As I understand it, bonding is safer than just pairing because with bonding the encryption keys are stored in the devices. In further communication it is not neccessary to exchange the keys and thereby expose them to MITM attacks. Are there any work being done to support createBond() also on iOS in order to achieve higher security?

pwespi, 2021-11-20:
Thank you for using this plugin.

On iOS bonding is handled by the OS. So if you interact with a characteristic of a peripheral device that requires encryption, iOS will automatically trigger the bonding process. The is no API available (to my knowledge) that can be used to trigger the bonding process by an app directly, therefore `createBond` cannot be implemented in the plugin for iOS.

closed 2021-11-20 by pwespi

aanders77, 2021-11-20:
That's great news, thanks :) PS! Maybe this information should be added to the readme?