capacitor-community/bluetooth-le #250: Unhandled Promise Rejection on iOS Simulator

maelp, 2021-11-22:
**Describe the bug**
I'm just running `BLEClient.initialize().then(() => bleAvailable=true).catch(() => bleAvailable = false)`, it works on the web and on iOS device, but on the iOS simulator, the call fails, but it also generate this error, and then the whole app "hangs" (router doesn't work, etc)

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Call the code above
2. Run it on the iOS simulator (I'm using iPhone 11, iOS 15)
3. Debug error using Safari > Develop > Simulator > (Choose your app)
4. See error

**Expected behavior**
The initialize should just throw an error which is caught in the `.catch` and not have the application stop working correctly

**Plugin version:**
- @capacitor-community/bluetooth-le: latest

**Desktop (please complete the following information):**
- OS: osX
- iOS simulator (iPhone 11, iOS 15)

**Smartphone (please complete the following information):**
- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

pwespi, 2021-11-22:
Thanks for using this plugin.

I cannot reproduce the behavior you're describing. Your line of code has a typo in `BleClient` and it's not clear what the variable `bleAvailable` is.

Please try the following:
```ts
BleClient.initialize()
  .then(() => {
    console.log("BLE available");
  })
  .catch(() => {
    console.log("BLE not available");
  });
```

With this code my app behaves as expected in the simulator and on the device.

maelp, 2021-11-22:
My exact code is in VueJS and Ionic

```
import { BleClient } from '@capacitor-community/bluetooth-le';
import { ref, Ref } from 'vue';

export const bleAvailable = ref(false);
export const bleInitialized = ref(false);
export const bleError: Ref<any|undefined> = ref(undefined);

export async function initBLE() {
    if (bleInitialized.value) {
        return;
    }

    try {
        await BleClient.initialize();
        bleAvailable.value = true;
        bleError.value = undefined;
    } catch (error) {
        console.error("initBLE", error);
        bleAvailable.value = false;
        bleError.value = (error as any);
    } finally {
        bleInitialized.value = true
    }
}
```

maelp, 2021-11-22:
Do you manage to have your code run on the iOS simulator?

pwespi, 2021-11-22:
Yes, my code runs on the iOS simulator and logs `BLE not available`.

The error `TypeError: No default value` suggests that you might be using a variable before it has been given a value.

pwespi, 2021-11-27:
Have you tried my code? Does it work?

maelp, 2021-11-28:
Ok for some reason now the code works :)
Thanks

closed 2021-11-29 by pwespi