capacitor-community/bluetooth-le #251: Can not connect after discover on Chrome

jt3u17, 2021-11-23:
connected to device {deviceId: 'ZuvwY/mRt69EhcAyFFmX7g==', name: 'BGX-2939', uuids: undefined}
ble.ts?033c:158 Error: Device not found. Call "requestDevice", "requestLEScan" or "getDevices" first.
    at BluetoothLeWeb.getDeviceFromMap (web.js?ca96:271)
    at BluetoothLeWeb.getCharacteristic (web.js?ca96:198)
    at BluetoothLeWeb.write (web.js?ca96:210)
    at eval (index.js?1547:196)
    at async BleClientClass.eval (bleClient.js?f44a:206)

I tried to connect to a BLE device and I am able to identify the device on Chrome but unable to pair. The normal deviceID should be a MAC address, however what I have obtained is ZuvwY/mRt69EhcAyFFmX7g==. I was wondering if this is the reason that I am not able to connect?

pwespi, 2021-11-27:
Thanks for using this plugin.

On Android the device ID is the MAC address, however on web and iOS the device ID is some other string (it's not possible to get the MAC address on web and iOS). So your example ID looks OK.

Does it only happen with this device, are any device? Can you share some code that reproduces the example? Have you tried on Android or iOS?

LgArcida, 2021-11-29:
Having a similar issue on WEB
Upon using `requestLEScan` it will only display the  confirmation to allow BLE scanning, never had the option to select a device from the list.
Using `requestDevice` does show all available devices and connection is successful, however the device appears to be without any service thus making it impossible to write/read.
Even after requesting the service with the device ID, none is given

pwespi, 2021-12-04:
> Having a similar issue on WEB Upon using `requestLEScan` it will only display the confirmation to allow BLE scanning, never had the option to select a device from the list. Using `requestDevice` [...]

On web, all services that you want to use have to be added to the `services` or `optionalServices` array in the `requestDevice` or `requestLEScan` call.

See:
* https://github.com/capacitor-community/bluetooth-le#usage
* https://github.com/capacitor-community/bluetooth-le#requestbledeviceoptions
* https://webbluetoothcg.github.io/web-bluetooth/#device-access-is-powerful

LgArcida, 2021-12-06:
Tried it as well.
Using both `services` and `optionalServices` but still I am not able to communicate with the device.
As said using `requestDevice` I was able to pair and connect to the device but since the `uuid` array is always empty then it not possible to write/read.
As far as I could see, the uuid list is added only when `requestLEScan` which have other issues on web

pwespi, 2022-01-10:
Is this still an issue? If yes, could you create a minimal reproduction and share the code?

closed 2022-02-10 by github-actions[bot]