capacitor-community/bluetooth-le #252: requestLEScan now working on web

LgArcida, 2021-11-29:
** WEB **

`requestLEScan` will only show the permission popup if there is a debugger on the code and will continue to do the same over and over again never saving the granted permission nor showing the available devices nearby.

**To Reproduce**
1. Open the web app with all the experimental flags enabled 
2. Trigger the `requestLEScan`
3. Notice how nothing will actually happen
4. Put a debugger and try again, permission popup displyed

**Desktop (please complete the following information):**
- OS: Ubuntu 20
- Browser Chrome
- Version 96

pwespi, 2021-12-04:
Thanks for reporting this.

`requestLEScan` is still experimental in Chrome (behind the `#enable-experimental-web-platform-features` flag) and can be buggy. What you're describing might be related to this bug in Chromium: https://bugs.chromium.org/p/chromium/issues/detail?id=1137504

I don't think there is much that can be done in the plugin until the underlying bugs in Chromium are fixed (except for improving the plugin documentation).

LgArcida, 2021-12-10:
Thanks, with requestDevice is working
Strange thing is that device name is undefined most of the time

pwespi, 2022-01-02:
Many devices do not advertise a device name. When I'm scanning I also get a lot of device where the name is undefined.

pwespi, 2022-01-10:
Is this still an issue?

jastax, 2022-01-22:
Yes

jastax, 2022-01-22:
I'm sorry, I answered here stating this was still an issue, but I think I'm running into another problem (requestLEScan callback ist not executed)

Thanks for your work

closed 2022-02-22 by github-actions[bot]

arigus, 2022-06-29:
RequestLEscan not working still. Permission allow closed automatically when clicking on the scan button. All demos on the net  are the same now