capacitor-community/bluetooth-le #259: Add connect / disconnect listener

aanders77, 2021-12-21:
**Is your feature request related to a problem? Please describe.**
I would like to show a status message to the user, describing if BLE is connected or not.

**Describe the solution you'd like**
I want to be able to register a listener which can trigger an event whenever the specified device connects/disconnects.

**Describe alternatives you've considered**
I've considered to show the status message based on programatically triggered events (connect(), disconnect()), but it will not work when the device disconnects for other reasons (distance, empty battery, etc).

**Additional context**
Status message example:
[image]

pwespi, 2021-12-23:
You can use the optional `onDisconnect` callback of the `connect` method.

E.g.
```ts
await BleClient.connect(device.deviceId, (deviceId) => {
  console.log(`device ${deviceId} disconnected`);
});
```

See https://github.com/capacitor-community/bluetooth-le#connect

aanders77, 2021-12-23:
I didn't notice that argument, thanks!

closed 2021-12-23 by pwespi