capacitor-community/bluetooth-le #260: Thread 1: EXC_BAD_ACCESS (code=1, address=0x10)

allidoisace, 2021-12-22:
**Describe the bug**
Thread 1: EXC_BAD_ACCESS (code=1, address=0x10)
That occurs at some point during subscription receiving data.

Device.swift
```
private func resolve(_ key: String, _ value: String) {
        let callback = self.callbackMap[key] // Thread 1: EXC_BAD_ACCESS (code=1, address=0x10)
        ...
```

**To Reproduce**
1. Subscribe to 4 characteristics.
2. 1 Characteristic is receiving data (< 200bytes) every 100ms.
3. 1 Characteristic is receiving data (< 200bytes) every 1000ms.
4. 2 other characteristics aren't receiving any data. 

**Expected behavior**
Expect the App not to crash when it appears to hit a null pointer in `resolve()`.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.5.0

**Desktop (please complete the following information):**
- N/A

**Smartphone (please complete the following information):**
- Device: iPad Pro
- OS: iOS 15.1

**Additional context**

pwespi, 2022-01-02:
Thank you for using this plugin. 

Unfortunately I'm not able to reproduce this error. Could you give me some more information? Does it happen immediately after subscribing to the 4 characteristics, or after some time? Does it always happen or just randomly? Can you reproduce it on other devices? Can you share the logs from Xcode?

allidoisace, 2022-01-05:
> Thank you for using this plugin.
>
> Unfortunately I'm not able to reproduce this error. Could you give me some more information? Does it happen immediately after subscribing to the 4 characteristics, [...]

Sorry, I thought I responded earlier. Yes, so I am able to hit the problem 100% of the time, but it isn't right away and not always after the same amount of time. It happens 100% of the time in less than 5mins. 

I have only tried it on the iPad Pro, but I can try on the iPhone.

What logs are you looking for, because XCode just stops logging and opens the debugger.

pwespi, 2022-01-07:
Thanks. The logs just before the `EXC_BAD_ACCESS` happens. To see if anything else apart from the notifications is going on.

pwespi, 2022-01-10:
I did some more testing, but was still not able to reproduce the `EXC_BAD_ACCESS`. I even went down to sending data every 25ms. 

Could you reproduce it on the iPhone?

Could you create a minimal reproduction and share the code?

allidoisace, 2022-01-11:
If I get the chance today, I will attempt reproduce on the iPhone. The logs prior don't look super interesting, just a lot of:

```
To Native ->  BluetoothLe removeListener 131651430
To Native ->  BluetoothLe addListener 131651431
To Native ->  BluetoothLe startNotifications 131651432
```

I will get some better logging today if possible.

allidoisace, 2022-01-14:
So, I was able to cause the issue on the iPhone as well. All messages are within the MTU size limit.

```
Set notifications true
Resolve setNotifications|<128-bit UUID>|<128-bit UUID> Successfully set notifications.
⚡️  TO JS undefined
Resolve callback not registered for key:  read|<128-bit UUID>|<128-bit UUID>
⚡️  TO JS {"value":"... "}
⚡️  To Native ->  BluetoothLe removeListener 54523565
⚡️  To Native ->  BluetoothLe addListener 54523566
⚡️  To Native ->  BluetoothLe startNotifications 54523567
Set notifications true
Resolve setNotifications|<128-bit UUID>|<128-bit UUID> Successfully set notifications.
⚡️  TO JS undefined
Resolve callback not registered for key:  read|<128-bit UUID>|<128-bit UUID>
⚡️  TO JS {"value":"..."}
Resolve callback not registered for key:  read|<128-bit UUID>|<128-bit UUID>
⚡️  TO JS {"value":"48 69 67 68 "}
⚡️  To Native ->  BluetoothLe removeListener 54523568
⚡️  To Native ->  BluetoothLe addListener 54523569
⚡️  To Native ->  BluetoothLe startNotifications 54523570
Set notifications true
Resolve callback not registered for key:  read|<128-bit UUID>|<128-bit UUID>
⚡️  TO JS {"value":"..."}
Resolve setNotifications|<128-bit UUID>|<128-bit UUID> Successfully set notifications.
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe removeListener 54523571
⚡️  To Native ->  BluetoothLe addListener 54523572
⚡️  To Native ->  BluetoothLe startNotifications 54523573
Set notifications true
Resolve setNotifications|<128-bit UUID>|<128-bit UUID> Successfully set notifications.
⚡️  TO JS undefined
⚡️  To Native ->  BluetoothLe removeListener 54523574
⚡️  To Native ->  BluetoothLe addListener 54523575
⚡️  To Native ->  BluetoothLe startNotifications 54523576
Set notifications true
(lldb)
```

Mind you these messages are flowing in like crazy. BLE is outputting this information every 100ms at the very least, which doesn't include the 4 other subscriptions. I'm not sure what it means to `removeListener`, `addListener` and then `startNotifications` for what seems to be every notification that comes from the device.

allidoisace, 2022-01-15:
Okay I figured out the issue. It was related to the `remove`, `add` of the listeners where my components was subscribing infinitely. However, I was under the assumption that subscribing to something that's already been subscribed to would just ignore it. 

I feel like I have worked with a few pub/sub models that work like this and maybe it would be a benefit for this plugin to work the same?

mutasimissa, 2022-01-15:
@allidoisace does that mean the problem was related to that your code running the start notification repeatedly, which naturally will cause a problem?
@pwespi is there a way to know that there is a current notification that is active and thus prevent creating another notifications thread?

allidoisace, 2022-01-17:
Yes, my code was running the start notification repeatedly.

pwespi, 2022-01-22:
Thanks for looking into this and reporting back. 

I agree that with some pub/sub models, you can resubscribe without side-effects. However I see `startNotifications` more like something similar to WebSockets, where you would also only open one socket per app and then share that data. Rewriting `startNotifcations` to allow resubscribing without side-effects would be quite some effort, so not sure if it's worth it. 

I added a note to the documentation. With this I would consider this issue closed for now. Let me know if you don't agree.

closed 2022-01-22 by pwespi