capacitor-community/bluetooth-le #270: BleClient.connect throws 'connection timeout' in 1.6.1

qpre, 2022-01-08:
**Describe the bug**
Hi,

when using 1.6.1, connect throws a connection timeout, even if connection succeeds.
reverting to 1.5.0 fixes the issue.

**To Reproduce**
```
try {
   await BleClient.connect(
      ble_device_id,
      (device_id) => console.log('disconnected', device_id)
   );
} catch (e) {
   console.error(e); // throws 'Connection Timeout'
}

try {
   // works like a charm 
   const value = await BleClient.read(
     ble_device_id,
     SERVICE_UUID,
     CHARACTERISTIC_UUID
   );

   console.log(ip);
} catch (e) {
   console.error(e);
}
```

**Expected behavior**
the error should not be thrown if connection was a success

**Plugin version:**
- @capacitor-community/bluetooth-le: 1.6.1 => fails
- @capacitor-community/bluetooth-le: 1.5.0 => works

**Smartphone (please complete the following information):**
- Device: iPhone 11
- OS: 15.2

qpre, 2022-01-08:
Turns out you need to stop scanning before you use BleClient.connect if you want to avoid the error.

closed 2022-01-08 by qpre

Benni77, 2022-08-16:
Hey Bro where did you get this information? Couldn't find anything about this. Is this based on try and error? Kinda experience something similar on my iPhone. 

> Turns out you need to stop scanning before you use BleClient.connect if you want to avoid the error.

✌️

qpre, 2022-08-16:
Hello @Benni77, this was concluded from trial and errors.