capacitor-community/bluetooth-le #271: configurable methods timeouts

qpre, 2022-01-09:
**Is your feature request related to a problem? Please describe.**
Currently, there is a timeout on write and read operations for android and iOS (set to 5 seconds)
it would be nice to have it configurable for long tasks running on the device.

**Describe the solution you'd like**
BleClient.read, BleClient.write should take a timeout option

pwespi, 2022-01-10:
Thank you for your feature request. I will consider this for the next version.

That being said, in my experience, long running tasks on the peripheral device are usually handled with a combination of write and notifications, e.g. a write starts a task and the result is sent back over a notification. This way the individual BLE operation can stay short.

qpre, 2022-01-11:
Thanks for your feedback, I did go with a write + notify implementation.
I think this is more useful as a way to have feature parity between web-ble and iOS/android implementations,
it took me a while to figure out why it was working in chrome and not in the mobile apps :) 

Thanks !

mutasimissa, 2022-01-11:
I agree with @pwespi that it's better to keep the connection short on the peripheral devices, but having a configurable timeout should be nice for some usecases.
Regarding writing + notify can @qpre share with us any code or pseudocode for an approach to do that, I want to implement that the right way in the react example I maintain.
Thanks!

qpre, 2022-01-12:
I don't have any shareable code right now but what I did was:

1 - on phone: start listening to notifications on the desired characteristic.
2- on device: keep a handler for this specific client to be notified
3- on phone: writeWithoutResponse to the desired characteristic
4- on device: start process of value, then use handler from 2 to send it back to phone
5- on phone: when value is notified, stop listening.
6- on device: remove notification handler from 2.

That's a bit hackish and only works if you have control on the device you are pairing with, if you don't I guess custom timeouts are needed :)

mutasimissa, 2022-01-15:
thanks for answering @qpre, but why you are stop receiving notifications ?
I don't think the current library implementation the notifications has no timeout, so why you don't keep it running and writeWithoutResponse multiple messages?

qpre, 2022-01-17:
I use this flow to replace long running reads or writes, I only expect one result per read/write, therefore I don't need to keep listening after receiving a result.

qpre, 2022-01-20:
Note: even for simple read values (no complex actions involved, just sending a 10 chars string over ble) 5 seconds is too low on old/low-end devices (where the same action in lightblue works fine on said device)

closed 2022-01-22 by pwespi

pwespi, 2022-01-22:
Thank you for you inputs. Configurable timeouts have been added in v1.7.0.

qpre, 2022-01-24:
Thanks a lot @pwespi !